prusa3d/PrusaSlicer #2313: Prusa Slicer 2.0.0 File Associations

GWdd, 2019-05-21:
Prusa Slicer 2.0.0+Win64
Windows 7 Pro x64
Prusa i3 MK3

### Behavior
After installation, file associations were left blank.  .STL was no longer associated to anything.

Attempting to associate .STL with Prusa-Slicer.exe failed. No reported reason.

I copied Prusa-Slicer.exe to  PrusaSlicer.exe.

Attempting to associate .STL with PrusaSlicer.exe succeeded.

Prusa-Slicer.exe does not appear in the "associable: programs list.
[image]
[image]

gysnst2, 2019-06-03:
I also see that my stl files will not associate to slicerprusa

rtyr, 2021-01-29:
I believe this no longer issue. Closing.

closed 2021-01-29 by rtyr