prusa3d/PrusaSlicer #2317: MMU file detection not working consistently

DesignFlaw06, 2019-05-21:
### Version
Version 2.0.0+win64

### Operating system type + version
Windows 10 64 Bit

### 3D printer brand / version + firmware version (if known)
MK3S 3.5.1 MMU2S 1.05

### Behavior
MMU files are not recognized when brought in to Prusa Slicer.  I have a 3MF file of a MMU model that worked fine in Slic3r PE.  Both with dragging the file in and doing a File Open.  Prusa Slicer doesn't seem to recognize it as a MMU and imports all objects separately.  Dragging all the STLs that make up the model also does the same thing.  It doesn't recognize that they are all part of the same model and treat them as such.  Slicer PE did this almost flawlessly. 

Workaround is to load one file and then add each file as a part under that.  That seems to create issues with being able to select extruders though.  That may be a separate issue that I need to investigate further. 

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
File is too big to upload here, but the 3MF file and the corresponding STL files can be located here. https://www.thingiverse.com/thing:3630177/files

FidelCapo, 2019-05-22:
Thank you for report. I can reproduce this issue.

Created JIRA-947

bubnikv, 2019-05-22:
This is a regression bug, where we decided to not ask that "multi-part" question for slicer's own project file. We did not correctly detected though whether the 3mf/amf files are slicer's project files or not. With 82740835fbe64af4bba11f0122706f0009a670be this has been rectified.

Closing. Thanks for the report.

closed 2019-05-22 by bubnikv