prusa3d/PrusaSlicer #2318: Selecting extruder for multiple parts not working properly

kagenokurei, 2019-05-21:
### Version
2.0.0+win64

### Operating system type + version
Windows 10 Pro x64

### Behavior
The UI offers the ability to select an extruder for multiple selected parts of a multi-part object. The extruder number is updated on the parts list window, but the colors in the 3D View stays with what they previously were. When sliced, the preview shows the previously extruder colors instead of the one configured in the parts list.

STR:
1. Load a multi-part model
1. Split into different parts
1. Multi-select a range of parts in the list view
1. Right click on the gear icon to the right of one of the selected parts in the list.
1. Select the option to select the extruder number for the selected parts
1. Change the extruder number for the selected parts via the dialog that appears
1. Verify that the extruder number for those parts have been updated
1. Observe that the 3D View did not reflect the change in extruder number for the selected parts
1. Slice the model
1. Observe that the preview also does not reflect the change in the extruder number.

[image]
In the image above, only 2 extruders are used. but we can see that the 3D View shows all 5 extruders being used.

#### Project File (.3MF) where problem occurs
https://1drv.ms/u/s!Aloe2xZ0MxNElq9Xv2tKJrbSU268Fg

FidelCapo, 2019-05-22:
Thank you for report. I can reproduce this issue.

Created JIRA-948

FidelCapo, 2019-05-28:
Thanks for report. This issue should be fixed in next release.
Closing.

closed 2019-05-28 by FidelCapo