prusa3d/PrusaSlicer #2321: [Feature Request] Extruder Color Picker

plorimer, 2019-05-21:
[image]
Rather than having to go into Printer Settings to change the extruder color to get a proper image of a multi-extruder print in the 3D view, it would be much more convenient if this could be accomplished on the main screen.

A couple of suggested ways to implement this:

In 3D editor view, right click on the filament to access a color-picker
Add a floating "tool" pallet to the upper left corner (similar to the one in the preview view). double clicking on any of the extruder colors here would bring up the color-picker
simply use the existing tool pallet in Preview to do the same.

bubnikv, 2019-05-22:
Clicking on the colors with the left mouse button works. Is it not sufficient?

[image]

plorimer, 2019-05-22:
Damn! I tried this prior to posting and it didn't work. The reason is I clicked on the color I wanted to change -  the right-most or the two, but you have to click on the left-most color to bring up the picker. clicking on the right color activates the pull-down menu to change the material type.

bubnikv, 2019-05-23:
> The reason is I clicked on the color I wanted to change - the right-most of the two, but you have to click on the left-most color to bring up the picker. 

That's a bug and we shall fix it.

@YuSanka please look into it. Thanks.

IntegersOfK, 2019-07-06:
@YuSanka @bubnikv Could you also please make it so when we select a different filament it will allow us to set the Extruder colour preview to that filament's default colour too? It's annoying that I have all my filaments inputted with proper colours but I still have to then go into the Extruder preview colour and set them there too.

Frustrating as described here:
https://www.reddit.com/r/prusa3d/comments/bexk08/slic3r_pe_1420beta2_color_change/el9j4ar

bubnikv, 2019-08-24:
The initial bug was fixed already.

The upcoming 2.1.0-beta will allow to reset the printer extruder color to "Undefined". In that case the real filament color will be used.

closed 2019-08-24 by bubnikv