prusa3d/PrusaSlicer #2322: Feature Request: Keyboard shortcuts help dialog should not be modal

davidhbrown, 2019-05-22:
### Version
Version 2.0.0+win64

### Operating system type + version
Windows 10 Pro, 1809

### 3D printer brand / version + firmware version (if known)
Irrelevant (but Original Prusa i3 Mk2.5 with firmware 3.7.1)

### Behavior
* _Describe the problem_
* As a new user, I would like to see the keyboard shortcuts onscreen while learning to work more efficiently, but the help dialog is modal, so I have to close it to actually use the shortcuts, then open it again, then close it again...
* _Steps needed to reproduce the problem_
* Select Help / Keyboard Shortcuts
* Click any control or input field or press any key (other than enter/escape); a warning beep is issued and the intended action is not performed.
* _Expected Results_
* I would like to be able to operate PrusaSlicer while simultaneously viewing the keyboard shortcuts
* _Actual Results_
* I have to close the keyboard shortcuts help to use PrusaSlicer

This is a request that an existing feature's behavior should be modified

#### Project File (.3MF) where problem occurs
No file is needed; this issue is apparent immediately after launching PrusaSlicer.

According to [the wxWidgets wxDialog Class Reference](https://docs.wxwidgets.org/3.0/classwx_dialog.html) changing this behavior might be as easy as changing a call to `ShowModal()` to `Show()`. Possible good first issue?

BerndJM, 2019-05-22:
As a quick workaround (until maybe it gets fixed):
Why not simply take a screenshot from the shortcuts window?

davidhbrown, 2019-05-22:
That is a good thought :-) I was having too much fun looking up the wxWidgets docs and trying to begin to understand the project structure to think of that!

bubnikv, 2019-05-22:
Sorry, but it works as intended. Keep in mind, that we are building an
application not just for geeks, but for beginners as well. You really want
to avoid the non-modal dialogs to be usable by a beginner. The beginner
certainly does not want to hunt for the windows. It is a better idea IMHO
to prepare a cheat sheet with keyboard short cuts to be printed.

davidhbrown, 2019-05-23:
I had not heard that "hunt for the windows" concern before; would be interested to read more. I'd been thinking of cautions about when **not** to use modals such as from [Nielsen Normal Group](https://www.nngroup.com/articles/modal-nonmodal-dialog/), [Google](https://material.io/design/components/dialogs.html#usage), and [Apple](https://developer.apple.com/design/human-interface-guidelines/ios/app-architecture/modality/). And of course, my own inconvenience in using this help screen. But if this is your intent, you're the developers. If the dialog included a link to such a printable cheat sheet, sure, that would be great... though I do have more room on my monitors than on my desk. Thanks for the reply.

bubnikv, 2019-05-23:
We are judging the simplicity for a first time user (imagine your kid or your grandma) versus power for the power users versus the load of our support. Making the feature configurable does not cut it in respect to our support load, as somebody will mess up the setup and then imagine the customer being on a phone line looking at a different screen than the support personnel.

davidhbrown, 2019-05-23:
Bubnikv, I appreciate your sharing your understanding and insights. However, I wonder if you might be thinking of a different issue? I didn't suggest making anything configurable. Just that you be able to leave the help visible on screen while continuing to use the program. 

Anyway, there are probably better ways to achieve the underlying goal -- for example, MS Office puts help information into a sidebar which is well suited to 16:9 monitors -- but switching from a modal to standard dialog seemed like it would probably be easiest to implement given the decision had already been made to put help information into a dialog.

(Should I close this or you?)

prowlmedia, 2019-09-17:
So... the modal pop for the keyboard shortcuts is an issue.  it doesn't always open in the foreground and can easily get lost behind other apps.  Then you go back to your platter and nothing works.   You can't actually use the app and look at the keyboard shortcuts at the same time.

bubnikv, 2019-09-17:
>  it doesn't always open in the foreground

What system? How can I reproduce it?

> can easily get lost behind other apps

What system? How can I reproduce it?

> Then you go back to your platter and nothing works. 

Again, how can I reproduce it?

> You can't actually use the app and look at the keyboard shortcuts at the same time.

No, the keyboard shortcuts dialog is modal.

BerndJM, 2019-09-17:
> You can't actually use the app and look at the keyboard shortcuts at the same time.
> > No, the keyboard shortcuts dialog is modal.

At least under Linux the modal keeps the focus so you can do nothing in the main window ...

davidhbrown, 2019-09-17:
I still think a modal dialog is exactly the wrong choice here for all the reasons mentioned in the links I gave previously (May 22), but under Windows, too, the dialog does keep focus. You can move the dialog away from the main window, even to a different monitor, which could be confusing. But I wasn't able to trick Windows into messing up the stacking order.

prowlmedia, 2019-09-17:
> No, the keyboard shortcuts dialog is modal.

Exactly. 

Mac or PC - a modal help window always a bad idea idea. You can’t explore or use the app at the same time.  Open it look for command ( load of which are missing like mirror x / y / z ) then 

On mac... What is quite annoying is that I used to be able to use the system MacOS keyboard shortcuts to assign ANYTHING in the menu bar to have my own shortcut.  

It’s one of the best things about MacOS - a unified Menu UI that is easily to customise.  Now that seems to have been disabled for some reason.

davidhbrown, 2026-09-18:
There is no keyboard shortcut help option in PrusaSlice 3.0.0 alpha, so either...

- The keyboard shortcut help screen feature is planned but not currently implemented and this issue may still be relevant and the issue should remain open
- The keyboard shortcut help screen feature is not planned and so this issue is indeed moot/legacy and the issue should be closed.

The general behavior of additional information screens in PrusaSlicer 3.0.0 alpha seems to be to create overlays that look like they ought to be dialog boxes except that you can't move them and they close if you click outside them. This would _not_ be an appropriate interface for a keyboard shortcut help screen that should remain visible for future reference and should be possible to resize and move out of the way while continuing to work.

I would also say that in my own experience, the PrusaSlicer 2 implementation of keyboard shortcut help was pretty much a failure. The only shortcuts I reliably know after 7 years (!) are "a" to auto-arrange and "f" to select a face.