prusa3d/PrusaSlicer #2323: Default install is C drive with no other choice

jwvaughn, 2019-05-22:
### Version
Version 2.0.0+win64

### Operating system type + version
Win Pro 64 bit Version 1809 OS build 17763.37

### 3D printer brand / version + firmware version (if known)
i3 MK3

### Behavior
The install neither asks for an install location nor does it honor the windows setting for where new apps will be saved.
The default install is to the C drive (an SSD in my case) an I prefer to install apps on D (an HD).

#### Project File (.3MF) where problem occurs
N/A

jwvaughn, 2019-05-22:
Update: Even though the installer said it was installing to C, it actually installed on the D drive.

closed 2019-05-22 by jwvaughn

bubnikv, 2019-05-22:
I passed your feedback to the guy who maintains the windows installer. Thanks for heads up.

Janys, 2019-05-22:
Hi jwvaughn, I prepared the latest Drivers & Apps installer - sorry to hear you have trouble with it.

A couple of things:
- If you're doing a clean install, the installer asks you for driver letter / path. If not, it means that you already have Prusa drivers installed somewhere in the system and the new ones will be installed into the existing folder. If you would like to install the driver package into a different location, please uninstall the obsolete version first - it's basically a safety check to make sure you don't have multiple Prusa packages in your system
- Some components (testing objects, Gcodes) are always installed onto the C drive, because they have to be placed in the Public Documents folder - so you may see the C drive path in the installer briefly. If you don't need the test objects, untick them in the components selection. No applications should be installed onto your C drive.

Please let us know if this makes sense, or if the problem seems to be elsewhere. If you could let us know what was installed on the C drive even though you selected the D drive, it would help us locate potential issues.

jwvaughn, 2019-05-22:
My thought is that it should always ask where to be installed.  I still consider this a bug even though it did install where I wanted it. However the initial indication was that it was going to the C drive. 

 

Thanks for the feedback.

Jerry

Janys, 2019-05-22:
> My thought is that it should always ask where to be installed. I still consider this a bug even though it did install where I wanted it. However the initial indication was that it was going to the C drive. Thanks for the feedback. Jerry 

Yes, thank you for the suggestion, I'll try to fix this for the next version - folder/drive selection will be always displayed

jwvaughn, 2019-05-22:
Thank you.

 

Jerry

KubaMiszcz, 2024-01-01:
yep, definitely, 
i have dir `/3Dprint`, and `/prusa slicer` inside (`/3Dprint/Prusaslicer`)
last time accidentally point to /3Dprint and installer deletes all content and install prusa slicer inside
luckily I recover files and dirs

but today I'll wait for choose install dir, click next, next, next and suddenly it starts to install and history repeats, it deletes all content and i recover only dirs, files in /3Dprint are gone

how it remembers this directory?

and definitely should always be input with point where it should be installed