prusa3d/PrusaSlicer #2327: The Slicer in Hebrew windows makes everything go right to left and won't work

shwekwek, 2019-05-22:
### Version
Version 2.0.0

### Operating system type + version
Windows 10 Home IN HEBREW (Arabic is also right to lefty language...)

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
everything is right to left, and not working. 
there is no visible control buttons to switch between sliced mode / objectmode. i cant delete or move an object. when i turn the view orientation, it goes to the other way (left/right). the white information box is on left of the screen. the name of the object is upside down (left / right). the hours time of print is erased/ hidden and not showing. and so on and on, everything looks wrong and not working.
[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-05-22:
That's like on an LSD trip.

Would you please try to go to the Config menu and switch to English? It
looks like you have the application set to Hebrew.

bubnikv, 2019-05-22:
To my reference:

we should change
bool GUI_App::load_language()
to check for the right-to-left flag and maybe reset it? Or force English
for now?

and here is a writeup on the right-to-left support in wxWidgets
https://www.wxwidgets.org/blog/2006/10/right-to-left-layout-rtl-support-in/

shwekwek, 2019-05-22:
nice one! that was quick. switched to English, everything looks great now! 
BUT, just FYI, im pretty sure that I chose English when requested to choose a language. Hebrew wasn't one of the options. (tbh I wasn't expecting Hebrew to be there, so that's fine).

bubnikv, 2019-05-22:
Let's keep it open, we shall avoid such a surprise.

bubnikv, 2019-05-22:
You selected English in the installer, but Slicer picked the language of your system.

shwekwek, 2019-05-22:
that's true. good point.

bubnikv, 2019-05-23:
By the way, what would you expect PrusaSlicer to look like if it was localized to Hebrew? Would you like to have all the controls mirrored as at the first screenshot you posted here? What is common for right-to-left localized applications? It was quite striking to me when I first saw it, but the wxWidgets guys (and seemingly the Linux / GTK guys as well) claim that it is the right thing.
https://www.wxwidgets.org/blog/2006/10/right-to-left-layout-rtl-support-in/

bubnikv, 2019-05-23:
Shall we just switch to English for the right-to-left languages for now?

shwekwek, 2019-05-23:
to me it's simple.
If the program is in Hebrew, and the command bar is in hebrew, everything should be from right to left, that's how i read, and that's how my logic work and how i expect the things to happen, from right to left. the operating system and the programs in hebrew work this way.
But, if the program is in english, it should work from left to
[image]
 right. i use obviously many apps and programs that are not in hebrew, and they are all from left to right. even though my system is in general in right to left, the other programs work normally with no interference. 
I'm attaching a prtscn shot of my desktop now. as you can see, the general layout of the os (windows 10) is from right to left. google chrome is in hebrew, and the layout is also right to left. solidworks is not in hebrew, so its left to right. until you asked me, i never noticed all that, so i guess we are used to it by now. the switching between languages as working between diff apps is a nightmare, tbh, but you get used to it too.

shwekwek, 2019-05-23:
I'm not sure if it's relevant, but if and when the time comes and you guys feel like translating the slicer to Hebrew, I would be more than happy to help from my side.

bubnikv, 2019-05-23:
Thanks for the hints. We will have hard time when somebody comes up with
the Hebrew translations :-)

bubnikv, 2019-08-31:
The upcoming PrusaSlicer 2.1.0-beta3 will detect a right-to-left system language and it will switch to English at first launch.

For now, we do not plan to make PrusaSlicer right-to-left capable. We will do that if there is a wider interest.

Closing.

closed 2019-08-31 by bubnikv

bubnikv, 2022-11-28:
@shwekwek Would you please get in touch with us per e-mail? 
vojtech.bubnik@prusa3d.cz
We need to consult right-to-left input for one of our newest feature to be released soon.
Thank you.