prusa3d/PrusaSlicer #2329: In MMU2 single mode, support filament selection from octoprint

lubarb, 2019-05-22:
### Version
2.0

### Operating system type + version
Windows 10 64

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3s & mmu2s

### Behavior
Running a mmu2s single mode print, the first step is to select the filament from the printer control panel.

When the print is initiated from Octoprint, you have to separately select the filament from the printer. It would be great if the printer & octoprint could coordinate so the filament could be selected from the octoprint control panel.

Second best would be if octoprint could show a status indicating if it is waiting for the printer filament selection.

New feature request

bubnikv, 2019-05-22:
Please open this issue at the firmware issue tracker. We have discussed the
synchronization between the OctoPrint and firmware internally, we even had
Gina for visit.

bubnikv, 2019-05-22:
Closing here in Slicer tracker, it shall be implemented as part of the firmware & OctoPrint (or Prusa specific OctoPrint plugin).

closed 2019-05-22 by bubnikv