prusa3d/PrusaSlicer #2330: Slicer does not import the entire object

PrusaSKuser, 2019-05-22:
PrusaSlicer 2.0.0

Windows 10, 64bit

3D printer brand / version + firmware version (if known)
Prusa i3 MK2 S - but this is not important

### Behavior:

The problem is, that Slicer reads/loads/imports only part of the 3D object created in Rhinoceros software. The file is in .stl format and I just want the gcode format from Slicer. Please see the screenshots. Slicer adds only the vertical part of the object, but does not "see" the horizontal part (the base).

Is there any option to solve this issue?

[image]
[image]

bubnikv, 2019-05-22:
you need to merge the two object into a single solid (use a boolean union)

PrusaSKuser, 2019-05-22:
Yes I was thinking similarly, but the object is just one (one connected surface) and there is nothing to merge with.. I cannot create a solid... I think as long as the object is one single object (though surface), Slicer should read it as one, but it does not...

PrusaSKuser, 2019-05-24:
Note: I have tried and created also different file formats (amf, 3mf) but the same problem with them as with stl. Slicer is not opening obj format files at all.
Anybody any advice with this..?

bubnikv, 2019-05-24:
you should produce a valid model.

rtyr, 2020-04-17:
Can you provide 3MF project file or stl so we can check? But as bubnikv said, it is most likely caused by invalid model.

rtyr, 2020-05-18:
No response. Closing.

closed 2020-05-18 by rtyr