prusa3d/PrusaSlicer #2333: Not-recommended use of miniz

jasontibbitts, 2019-05-22:
### Version
2.0.0

### Operating system type + version
Fedora 31

While looking at using the system version of miniz (which isn't difficult at all) I noticed that PrusaSlicer's use of the bundled miniz code runs counter to what upstream suggests.  Many of the upstream files have been stripped but if you go to the original repository (https://github.com/richgel999/miniz), you see this:

```
Please use the files from the releases page in your projects. Do not use the git checkout directly! The different source and header files are amalgamated into one miniz.c/miniz.h pair in a build step (amalgamate.sh). Include miniz.c and miniz.h in your project to use Miniz.
```

But the PrusaSlicer code includes the original source header files instead of the amalgamated miniz.h.  I've patched a few files to just include `<miniz.h>` and everything builds fine with the system version but I don't have a good way to test all of the potential effects of that.  (Given how many security issues have been caused by bugs in archivers and decompression programs, we want to avoid bundling them as much as possible.)  And of course doing this means you must use the system version, so I didn't think a PR would be useful.

bubnikv, 2019-05-23:
I have asked @tamasmeszaros to look into this issue. As you noted, we will need to retest all the code that uses the miniz library after the change.

tamasmeszaros, 2019-05-23:
@jasontibbitts: I've updated the bundled version of the library as you have suggested. Unfortunately, I haven't found miniz packages in Ubuntu repositories which is installed on my dev machine. I see that Fedora has the packages for miniz. For now, we will test the app with the new version bundled. According to our future capacities, we may provide a cmake find module for miniz to use the system version on Fedora.

closed 2026-09-29 by prusaslicer-bot[bot]