prusa3d/PrusaSlicer #2334: Disable slicer firmware changes

Jays3DAdventure, 2019-05-22:
### Version 

Version 2.0.0+Win64

### Operating system type + version

Windows 10 64bit

### 3D printer brand / version + firmware version (if known)

Custom D-bot + Marlin 1.9

### Behavior

_Is this a new feature request?_

Please add an option to turn off the slicer changing the firmware setting.

#### Project File (.3MF) where problem occurs

bubnikv, 2019-05-23:
I don't understand what you mean.

Jays3DAdventure, 2019-05-23:
With a custom printer in prusa slicer, I do not want the slicer changing firmware settings, M201, M203 etc.

I want an option when adding custom printers to disable this behaviour.

bubnikv, 2019-05-23:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1212

closed 2019-05-23 by bubnikv