prusa3d/PrusaSlicer #2335: Lost Bed image in Slicer 2.0.0

shooterlu, 2019-05-22:
### Version
Version 2.0.0+win64

### Operating system type + version
Win 10 Pro x64

### 3D printer brand / version + firmware version (if known)
Prusa I3 Mk2.5 
Firmware 3.7.1

### Behavior
When i Change the Language i lost the Image of the Bed in PrusaSlicer.
I only have the grid like the old Slicer.
[image]

cchanus, 2019-05-22:
I having a similar issue but I don't even get the image of the bed in PrusaSlicer.
The language is default in English
[image]

bubnikv, 2019-05-23:
What is your GPU? Maybe you are low on GPU memory and your issue is similar to https://github.com/prusa3d/PrusaSlicer/issues/2299 ?

Would you pleas retest with the following build, that uses GPU texture compression?

[PrusaSlicer-2.0.0+11-et-experiments-win64-gefd247fc5-201905211220.zip](https://github.com/prusa3d/PrusaSlicer/files/3210793/PrusaSlicer-2.0.0%2B11-et-experiments-win64-gefd247fc5-201905211220.zip)

What printer profile is active? The system one by PrusaResearch, or yours?

shooterlu, 2019-05-23:
It was mine.
So when i took back the original Prusa Profile an restart PruseSlicer, the plate is back. When i took mine and restard the Slicer i have the grid.

Thx for your Help

bubnikv, 2019-05-23:
For some reason the link from your profile to our system profile was lost.
It is the link to our system profile, which makes the texture bed to be
displayed.

This is what you shall see for your custom profile, if you derived it from
ours.

[image: image.png]

shooterlu, 2019-05-23:
Yes, but when i make changes (adress for octoprint for example) an do a copy of the original profile, then the bed ist lost again.

shooterlu, 2019-05-23:
Yes know it works when i make a copy
Thx

closed 2019-05-23 by shooterlu

bubnikv, 2019-05-23:
>  Yes, but when i make changes (adress for octoprint for example) an do a
copy of the original profile, then the bed ist lost again.

I just verified it works. Would you please confirm it? Thanks.

shooterlu, 2019-05-23:
Yes it works. Thx