prusa3d/PrusaSlicer #2336: Crash on export Config Bundle

ArcRahl, 2019-05-22:
### Version
_2.0.0+win64_

### Operating system type + version
_Windows 10 v1803_

### 3D printer brand / version + firmware version (if known)
_Original Prusa i3 MK3_

### Behavior
When I try to export Config bundle, PrusaSlicer shuts down/crashes. It exports a small (18 KB) file. When I try to import the file into PrusaSlicer on another computer, I get this error:
[image]

ArcRahl, 2019-05-22:
Here is the .ini file it exports before it crashes:
[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/3208581/PrusaSlicer_config_bundle.zip)

bubnikv, 2019-05-23:
This is a duplicate of #2307, it has already been fixed, the fix will be part of the next minor release.
It will work if you install the SL1 printer and select it at least once.

closed 2019-05-23 by bubnikv