prusa3d/PrusaSlicer #2337: [Feature Request] Allow to use wipe tower for non MMU printers.

Denarian, 2019-05-22:
### Version

2.0

### Operating system type + version

Windows 10 1803

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 stock
Custom hypercube + diamond hotend - Marlin Firmware (in assembly)

_Is this a new feature request?_
YES

For now wipe tower is activated only if we select "Single Extruder Multi Material" but this also activates MMU filament change routine and tool change Gcode is not used. Solution for that is simply create separate checkboxes for "Single Extruder Multi Material" and "MMU"

lukasmatena, 2019-05-22:
There is bit more work than "simply create checkbox", but hopefully we'll get there soon. @tcm0116 already looked into it and opened pull request https://github.com/prusa3d/PrusaSlicer/pull/2198. I hope that after the PrusaSlicer 2.0.0 release settles down a bit, we'll have time to integrate the contribution.

Denarian, 2019-05-22:
I know. It was just an idea how it's can look like from user point of view. It will require changing some program logic. I can help with implementation but I'm too busy with my own programs :P

tcm0116, 2019-05-23:
@Denarian - I also have a printer with a Diamond hotend, which is why I'm working on this change. One thing I've learned is that the optimized wipe tower is probably over optimized for a mixing extruder. As it currently works, if the print doesn't need a tool change for several layers, then the wipe tower will not perform any tool changes until it needs to. This is great for a switching style extruder, but in a mixing extruder the unused filaments can get jammed from sitting unused after a long period. I'm thinking about adding a setting to the wipe tower to force it to extrude each filament into the wipe tower even if a tool change is unneeded. I have the logic worked out, but I just need to add the setting.

lukasmatena, 2019-12-17:
The wipe tower can be generated for non single extruder MM printers starting with PrusaSlicer 2.1.0. Ramming etc. is not performed in that case, further customization is possible by custom toolchange gcode, which is used even with the wipe tower.

I think this issue can be closed. The Diamond hotend issue mentioned by @tcm0116 is IMHO independent and should be discussed in #3044.

closed 2019-12-17 by lukasmatena