prusa3d/PrusaSlicer #2338: [Request] fan

duartemv, 2019-05-22:
### Version 2.0.0+win64

### Windows 10 Professional

### Prusa i3 MK3 + firmware version 3.5.1 7x7 (user made)

### Behavior

Is this a new feature request? YES

In FILAMENT SETTINGS > COOLING it is possible to disable the fan for the first XX layers.

On some prints, I need to have the fan always running after layer 3 or 4, and then disconnect at a height of ~10mm, or after 50 layers (for 0.2mm layers).

When the option "KEEP FAN ALWAYS ON" is enabled, there should be an option to disconnect the fan after YY layers, or Zmm height.
Something like the attached image below.

[image]

Current procedure: I use octoprint to disable the fan manually and lock it so the gcode doesn't turn it back on for the rest of the print.

bubnikv, 2019-05-23:
You can do that using a post-processing script called from the slicer
https://hackaday.com/2016/07/20/3d-printering-g-code-post-processing-with-perl/

I know Simplify3D suports some very simple G-code processing language (find & delete or replace), maybe we can add such a functionality to cover the needs of majority of tinkerers.

rtyr, 2021-01-29:
Basically same as https://github.com/prusa3d/PrusaSlicer/issues/3439, https://github.com/prusa3d/PrusaSlicer/issues/3635. Closing this one.

closed 2021-01-29 by rtyr