prusa3d/PrusaSlicer #2339: Jerk limits use incorrect units

DominikPalo, 2019-05-22:
### Version
2.0.0+

### Operating system type + version
macOS Moje 10.14.4

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
1) Go to "Printers Settings" -> "Machine limits" -> "Jerk limits" (in Advanced mode)
2) Observe, that jerk limits use mm/s units although the correct units for jerk are mm/s^3 (see https://en.wikipedia.org/wiki/Jerk_(physics))

[image]

Sebastianv650, 2019-05-22:
mm/s is correct. Jerk isn't defined this way in Marlin & PrusaFW.
http://marlinfw.org/docs/configuration/configuration.html#jerk-

DominikPalo, 2019-05-22:
@Sebastianv650 ah, thanks! Maybe would be useful to display a tooltip over the jerk settings with that explanation:
> Jerk works in conjunction with acceleration (see above). Jerk is the maximum change in velocity (in mm/sec) that can occur instantaneously. It can also be thought of as the minimum change in velocity that will be done as an accelerated (not instantaneous) move.

Promite2501, 2019-05-23:
Yeah I'm with @DominikPalo, it always bugs me that the units in most slicer programs and firmware are wrong. People already struggle to get there heads around Jerk.    mm/s^3 
Its not really a feature of a bug though, just a common misconception.

velocity  = speed with direction  = mm/s
acceleration = rate of change in velocity = mm/s^2
jerk = rate of change in acceleration = mm/s^3

Sebastianv650, 2019-05-23:
@Promite2501 

> People already struggle to get there heads around Jerk.

You name it.
The unit isn't wrong as it isn't jerk in it's real form. But while "Some value in mm/s which isn't jerk but acts as jerk limitation" would have been a more honest feature name, the developers would be drowning in issues as the users would have no clue what's going on. So, name it jerk and live in peace ;-)
If you dig into the jerk code and calculate some examples, you will see that even the description from Marlin is only halve the truth. But that doesn't matter, it works in reality in most cases.

If you want something better, but as the issue tracker entries are showing even more not-user-friendly, read on over junction deviation which replaces jerk in Marlin. Now you have a unit in mm for jerk limitation. If you struggled with mm/s, you will have a happy day :) 

BTW this all has nothing to do with Prusa Slicer so I think it should be closed. It's all about motion control inside the firmware.

bubnikv, 2019-05-23:
The GRBL project overloaded the term of "jerk" to define maximum instantaneous velocity change, which is easier to be implemented by 8bit controllers, and most of today's 3D printer firmwares are based on the GRBL planner

I have added a note to the FAQ page.
Closing.

bubnikv, 2019-05-23:
https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)

closed 2019-05-23 by bubnikv