prusa3d/PrusaSlicer #2341: "a toolpath outside the print was detected" error  when "enable Prime all printing extruders" option active

marcottt, 2019-05-22:
### Version
2.0.0 win64

### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
anycubic kossel linear plus, marlin 1.1.9
dual material mod (2 motor, 1 nozzle)

### Behavior
configure 2 extruder with 1 nozzle
import stl dual material
enable prime tower
enable "Prime all printing extruders" option

prusaslicer create an artefact from outside bed to bed... unable to slice

_Is this a new feature request?  NO

#### Project File (.3MF) where problem occurs

[dual_extrusion_calibration_cube_5_part_1.zip](https://github.com/prusa3d/PrusaSlicer/files/3209301/dual_extrusion_calibration_cube_5_part_1.zip)

bubnikv, 2019-05-23:
You need to disable

[image]

It's clear that the priming areas do not quite work for circular beds, at minimum they need to be moved to the side from the center.

lukasmatena, 2019-11-19:
The positioning of the priming sections on circular beds has been improved with c84b1ca34b86e4f35b6caac1a8e51897951f14de, which is part of PrusaSlicer 2.1.0 already. More details also in #1121. That should solve this issue for good.

Closing.

closed 2019-11-19 by lukasmatena