prusa3d/PrusaSlicer #2343: "SegFault"

pci06, 2019-05-22:
### Version
2.0

2.0.0+linux64

### Operating system type + version
=>lsb_release -a ; echo "Kernel: " $(uname -a)
LSB Version:	core-9.20170808ubuntu1-noarch:security-9.20170808ubuntu1-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic
Kernel:  Linux thinkpad.local 5.1.3 #1 SMP Thu May 16 22:40:48 CEST 2019 x86_64 x86_64 x86_64 GNU/Linux

OpenGL installation
GL version: 3.1 Mesa 18.2.8
Vendor: nouveau
Renderer: NV117
GLSL version: 1.40
Installed extensions:
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_performance_monitor
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_trinary_minmax
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_query_buffer_object
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_ballot
GL_ARB_shader_bit_encoding
GL_ARB_shader_clock
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_image_samples
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_draw_buffers
GL_ATI_fragment_shader
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_texture
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_gpu_program_parameters
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_latc
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_EXT_window_rectangles
GL_IBM_multimode_draw_arrays
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_KHR_blend_equation_advanced
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_texture_compression_astc_ldr
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_NV_vdpau_interop
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* PrusaSlicer crash
* _Steps needed to reproduce the problem_
1) run PrusaSlicer
2) Add an STL file
3) Right click on the newly added object
4) Select and click on Extruders
5) Go to the top most right panel and click on "Extruders"
6) Press [Del] key

* _Expected Results_
Extruder item removed
* _Actual Results_
Crash
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

lukasmatena, 2019-05-22:
Thanks for the report. I got backtrace from ASAN on my system. I believe it will shed some light on what's rotten.

==19529==ERROR: AddressSanitizer: heap-use-after-free on address 0x61100090bc08 at pc 0x000001127de9 bp 0x7ffff7e84e10 sp 0x7ffff7e84e00
READ of size 4 at 0x61100090bc08 thread T0
    #0 0x1127de8 in ObjectDataViewModel::GetParent(wxDataViewItem const&) const (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x1127de8)
    #1 0x1532eb0 in wxDataViewCtrlBase::ExpandAncestors(wxDataViewItem const&) src/common/datavcmn.cpp:1244
    #2 0x154916f in wxDataViewCtrl::OnInternalIdle() src/gtk/dataview.cpp:4737
    #3 0x1687fc8 in wxWindowBase::SendIdleEvents(wxIdleEvent&) src/common/wincmn.cpp:2828
    #4 0x1688017 in wxWindowBase::SendIdleEvents(wxIdleEvent&) src/common/wincmn.cpp:2844
    #5 0x1688017 in wxWindowBase::SendIdleEvents(wxIdleEvent&) src/common/wincmn.cpp:2844
    #6 0x1688017 in wxWindowBase::SendIdleEvents(wxIdleEvent&) src/common/wincmn.cpp:2844
    #7 0x1688017 in wxWindowBase::SendIdleEvents(wxIdleEvent&) src/common/wincmn.cpp:2844
    #8 0x1688017 in wxWindowBase::SendIdleEvents(wxIdleEvent&) src/common/wincmn.cpp:2844
    #9 0x15c30ae in wxFrame::SendIdleEvents(wxIdleEvent&) src/gtk/frame.cpp:165
    #10 0x15e5906 in wxAppBase::ProcessIdle() src/common/appcmn.cpp:406
    #11 0x156b9c8 in wxApp::DoIdle() src/gtk/app.cpp:152
    #12 0x156bacf in wxapp_idle_callback src/gtk/app.cpp:100
    #13 0x7f80c263f049 in g_main_context_dispatch (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a049)
    #14 0x7f80c263f3ef  (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a3ef)
    #15 0x7f80c263f711 in g_main_loop_run (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a711)
    #16 0x7f80c3e6b696 in gtk_main (/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0+0x130696)
    #17 0x16cc154 in wxGUIEventLoop::DoRun() src/gtk/evtloop.cpp:65
    #18 0x1768d93 in wxEventLoopBase::Run() src/common/evtloopcmn.cpp:76
    #19 0x173ad10 in wxAppConsoleBase::MainLoop() src/common/appbase.cpp:380
    #20 0x17b3613 in wxEntry(int&, wchar_t**) src/common/init.cpp:507
    #21 0x84a1a1 in Slic3r::CLI::run(int, char**) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x84a1a1)
    #22 0x8192f3 in main (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x8192f3)
    #23 0x7f80c14ae82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #24 0x8428e8 in _start (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x8428e8)

0x61100090bc08 is located 136 bytes inside of 256-byte region [0x61100090bb80,0x61100090bc80)
freed by thread T0 here:
    #0 0x7f80c53ccb2a in operator delete(void*) (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x99b2a)
    #1 0x1140e38 in ObjectDataViewModel::Delete(wxDataViewItem const&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x1140e38)
    #2 0x1021987 in Slic3r::GUI::ObjectList::del_subobject_item(wxDataViewItem&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x1021987)
    #3 0x102bcda in Slic3r::GUI::ObjectList::remove() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x102bcda)
    #4 0x102c047 in Slic3r::GUI::ObjectList::key_event(wxKeyEvent&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x102c047)
    #5 0x1836bb3 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) src/common/event.cpp:1396

previously allocated by thread T0 here:
    #0 0x7f80c53cc532 in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x99532)
    #1 0x11422c0 in ObjectDataViewModel::AddSettingsChild(wxDataViewItem const&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x11422c0)
    #2 0x10286b2 in Slic3r::GUI::ObjectList::update_settings_item() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x10286b2)
    #3 0x1028a57 in Slic3r::GUI::ObjectList::get_freq_settings_choice(wxString const&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x1028a57)
    #4 0x1028ea7 in std::_Function_handler<void (wxCommandEvent&), Slic3r::GUI::ObjectList::create_freq_settings_popupmenu(wxMenu*)::{lambda(wxCommandEvent&)#1}>::_M_invoke(std::_Any_data const&, wxCommandEvent&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x1028ea7)
    #5 0x1836bb3 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) src/common/event.cpp:1396

SUMMARY: AddressSanitizer: heap-use-after-free ??:0 ObjectDataViewModel::GetParent(wxDataViewItem const&) const

bubnikv, 2019-08-01:
The following commit 93f86b795d048e5368986dbbdf51e898a841f14c adds asserts to analyze this issue. @lukasmatena could you still reproduce the issue?

lukasmatena, 2019-08-01:
I can. All I know so far is that this is Linux-specific issue, Win and OSX seem to work fine.

Now it trips on assert(m_idx >= -1) in ObjectDataViewModelNode::valid(). Top of call stack looks like this:

0  0x0000000003644d0e  ObjectDataViewModelNode::valid  /home/lukas/Slic3r/Slic3r/src/slic3r/GUI/wxExtensions.cpp  503
1  0x0000000003655e07  ObjectDataViewModelNode::GetParent  /home/lukas/Slic3r/Slic3r/src/slic3r/GUI/wxExtensions.hpp  267
2  0x00000000036490f7  ObjectDataViewModel::GetParent  /home/lukas/Slic3r/Slic3r/src/slic3r/GUI/wxExtensions.cpp  1435
3  0x0000000003952bd6  wxDataViewCtrlBase::ExpandAncestors  /home/lukas/Slic3r/Slic3r/deps/build/dep_wxwidgets-prefix/src/dep_wxwidgets/src/common/datavcmn.cpp  1248

lukasmatena, 2019-08-06:
Thanks to @bubnikv's investigation, the issue is fixed by https://github.com/prusa3d/PrusaSlicer/commit/d36c64873d9260c26be552905a4c2edb10256f79. The fix will be part of the upcoming PrusaSlicer2.1.0-alpha1 (due in several days). I'm closing the issue, thanks for the report.

closed 2019-08-06 by lukasmatena