prusa3d/PrusaSlicer #2344: Object teleports off platter when attempting to move it

ddthj, 2019-05-22:
Slic3r Prusa Edition
Version: 1.41.3+win64
Build:   Slic3rPE-1.41.3+win64-full-201902111723
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: 
Microsoft Windows [Version 10.0.17134.765]

### Behavior
Problem: When trying to click+drag an object on the platter in 3D view, the object teleports off the print bed and cannot be returned unless "Arrange" is pressed.

Steps:
1. Load any model onto the platter
2. Select a printer that has a large bed (I'm using a 1219x300mm bed
3. Try and click+drag part

Video of problem:
https://i.gyazo.com/1f852b024ab0652f1f7dda54accf42b0.mp4

lukasmatena, 2019-05-22:
Please, retest with the fresh release PrusaSlicer 2.0.0 and close the issue if the problem no longer appears (which will almost certainly be the case).

ddthj, 2019-05-23:
Yup that worked, Sorry for the trouble.

closed 2019-05-23 by ddthj