prusa3d/PrusaSlicer #2346: [Feature Request] Color Change preview colour picker

tubbers, 2019-05-23:
Version 2.0

This is a new feature request.

Love the new colour change fuctionality in PrusaSlicer v2 https://blog.prusaprinters.org/prusaslicer-2-release/#10

The new multi colour preview is super helpful too. Have messed up so many with incorrect layer number. The colours in the preview are a rainbow, starting from bottom to top. Would be great to have a colour picker to be able to select the preview colour for each colour change, so that the user can absolutely confirm the colours are right. If the user picked from a standard palette, it would be even possible for the Prusa printer LCD display to ask the the new colour by name "Please insert 'red' filament and press to load"!

JohnnyDeer, 2020-04-01:
Hi, in newest version is possible to pick color for each colour change.

tubbers, 2020-04-15:
Great - have tested it and love it. One question - i dont seem to be able to change the colour of the first layer from pink. I can change all the others, just not the bottom one.

JohnnyDeer, 2020-04-15:
Nice to hear, that you like it. You can change first layer color in side panel, where you choose filament type, just by clicking on color square. If you are satisfied, than please close this issue.

tubbers, 2020-04-18:
great

closed 2020-04-18 by tubbers