prusa3d/PrusaSlicer #2347: Missing the Layer height table from the Object Settings menu

udif, 2019-05-23:
### Version
2.0.0

### Operating system type + version
WIndows 10 Pro, 1809.

### Behavior
In Slic3rPE Prusa Edition 1.4.3 and earlier, I could select the object , go to settings, and selecty the "Layers" tab, where I could also control the layer height in a table, in addition to the slider from the "Layer editing" button. The slider would modify this table, but could be reset back to the table values.

In 2.0.0, I no longer find a table. The other settings have moved to a sidebar, but selecting "Layers and Perimeters" only leads to generic top/button/permieter parameters, not the layer height table.

### Expected Behavior
Add back the old table based layer height from 1.4.3
Without a table, setting layer height by hand is not accurate or repeatable.

tkircher, 2019-05-23:
Yeah, the paint-style editor can't be used to make precise changes in layer height. I thought it might be possible to use modifiers in v2 to effect layer height changes, but I tried that and it doesn't work.

bubnikv, 2019-05-23:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/2276, please read the comments there.

The problem with the modifier meshes and the layer height is, that the modifier would have to span the complete layer to work reliably. 

We plan to add something like the slabs, but where you would define the min Z, max Z, layer height and all the usual attributes you can apply to modifiers.

tkircher, 2019-05-23:
Hopefully sooner than later. I need to be able to specify, for example, exactly 0.2mm layers to 5.2mm, then 0.1mm layers up to 11.6mm, then 0.2mm layers to 20mm. It has to be exact with sharp changes, not smoothly varying layer heights and approximate areas. Which is why I still have to use an old version when I want to have different layer heights.

udif, 2019-05-23:
@bubnikv, Obviously this should not be a global setting, not for each object. I never bothered learning how this worked in practice because for the few times I needed that, I always had one object.

tkircher, 2019-05-23:
As long as extruder clearance is specified, it wouldn't be too complicated to plan for multiple objects each with different layer heights. Especially since the 'complete individual objects' option still exists.

rongith, 2019-05-23:
The same here, I use 3 versions of the Prusa slicer. 1.41.3 to specify layer height, supermerill fork for ironing and the official one for other applications.

Mithrandil, 2019-05-31:
They could use a function similar to the ColorPrint, but instead allow to set the layer height for every color
[image]

jimmy00784, 2019-05-31:
In the absence of now unusable feature that allowed modifying the layer height through modifier, another feature is needed to allow reliably set layer heights change that is precise, unlike the variable height feature. I like the suggestion by @Mithrandil but I'd also be ok if layer height modifier is re-enabled with additional checks to ensure it spans entire layer. 
I still like the old approach which allowed tweaking layer heights for specific sections. The final result might not be clean but a good enough for prototype prints with varying levels of detail.

bubnikv, 2019-08-01:
The feature will be resurrected in the upcoming PrusaSlicer 2.1.0-alpha1, allowing not only layer height to be defined for a Z range, but also other modifier mesh parameters.

closed 2019-08-01 by bubnikv

robscar, 2019-12-10:
have not seen this in 21.0. Is it implemented and I have just overlooked it?

robscar, 2019-12-10:
just to be sure we talk about the same missing feature, I want to print an object of 5.7mm height. Up to 5.6 using 0.2 and the last layer in 0.1. FYI, because of the object design I can't just print the first layer in 0.3mm

rongith, 2019-12-10:
@robscar
Yes it is, right click the object and then "Height range modifier"

robscar, 2019-12-10:
i'm impressed, looks great, directly test printing.  And thank you for the super fast response :-)