prusa3d/PrusaSlicer #2348: PrusaSlicer exits after entering illegal object name

jpcuzzourt, 2019-05-23:
### Version
PrusaSlicer 2.0.0+win64

### Operating system type + version
Windows 10 (build 17763), 64-bit edition

### 3D printer brand / version + firmware version (if known)
N/A
### Behavior
Prusa crashes if I accidentally rename an object in the object browser with illegal character,
and then navigate to another object before correcting the error.
* _Describe the problem_
* _Steps needed to reproduce the problem_
1. Load 2 or more objects.
2. Highlight one of them in object browser
3. Left click to rename it and type "foo/bar". Do not press Enter.
4. Now click on the name of object 2.
[image]

* _Expected Results_
An error message telling me name contains illegal characters
* _Actual Results_
An error message telling me name contains illegal characters, followed by application exiting unceromoniously.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

[My_Spool_Prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/3211233/My_Spool_Prusa.zip)

bubnikv, 2019-05-23:
thanks for the report, I can reproduce it.

bubnikv, 2019-05-23:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/666e9687ce16e386d218692d01eef6979fb9bd69
Thanks for the report.

closed 2019-05-23 by bubnikv