prusa3d/PrusaSlicer #2349: [year] in output filename format results in blank file name

okebase, 2019-05-23:
Version 2.0.0 Win64

Windows 10 Home
10.0.17134

MK3S stock firmware 3.7.0

this line in output filename format: 

{input_filename_base}_[year].gcode

produces a blank filename when I export

[month] and [day] work as expected

BerndJM, 2019-05-23:
strange ;-))
in the Linux version the same output filname format results in something like this:
Y-belt_holder-x:100% y:100% z:100%.gcode

Grismo, 2019-05-23:
Just noted the same with version 2.0.0 Win64 under Windows 7 Professional. I'm using the sequence: [input_filename_base]_[filament_preset]_[print_preset]_h[layer_height]_in[fill_density]-[fill_pattern]_p[perimeters]_sup-[support_material]_{filament_type[0]}_[year]-[month]-[day].gcode
That used to work without problems under Slic3r PE latest version. Now it produces a blank filename as okebase reported. By removing just the part [year] it works like intended.

epeleg, 2019-06-25:
Is there a link in to show what fields can be used in this?
and why for example filament_type needs a [0] after it?

neslekkim, 2019-10-21:
Still, Year does not work. month, day, hour, minute, second works, but year gives blank filename.
timestamp can be used, but would be nice if there was an "datestamp" :),, a timestamp with only the datepart.

Jebtrix, 2019-10-22:
This problem is in 
* File: **PrintBase.cpp**
* Function: `void PrintBase::update_object_placeholders(DynamicConfig &config, const std::string &default_ext) const`

...
` config.set_key_value("year", new ConfigOptionStrings(v_scale));`
...

@bubnikv The year is being changed to scale. Probably just in testing but forgot to change back?

lukasmatena, 2020-01-31:
Fixed by @Jebtrix in #3136 - thanks. The fix will be part of the next release 2.2.0-alpha3.
Closing the issue.

closed 2020-01-31 by lukasmatena