prusa3d/PrusaSlicer #2352: Cant control solid infill

userosos, 2019-05-23:
Hi! What i can control the value of solid infill. There are a lot of solid infill. Solid infill threshold area not working for me.

[image]

Version of PrusaSlicer V 2.0.0
Windows 10 enterprice 64 bit

[emblema_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/3212832/emblema_fixed.zip)

bubnikv, 2019-05-23:
I don't quite understand your problem. Solid infill is always 100%.

userosos, 2019-05-23:
> 
> I don't quite understand your problem. Solid infill is always 100%.
> […](#)

I can not control it. It quantity very big.

bubnikv, 2019-05-23:
I think I am starting to understand your problem. You believe that
PrusaSlicer puts too much of solid infill inside the object, right? Then it
is a duplicate of  https://github.com/prusa3d/PrusaSlicer/issues/1054 ,
right?

Or do you complain, that the solid infill is too solid?

userosos, 2019-05-23:
> I think I am starting to understand your problem. You believe that PrusaSlicer puts too much of solid infill inside the object, right? Then it is a duplicate of #1054 , right?

Oh. yes. It is duplicate. I can fix it now?

> Or do you complain, that the solid infill is too solid?

If i will can edit the parameter It will be good. I will st to 0% it.

userosos, 2019-05-23:
I change fill patern from stars to Line and it will OK.
[image]

userosos, 2019-05-23:
I can influence to the parametr use Solid layers - Bottom (i set it to 10). If i set to 5 i can see:
[image]
It looks like good. But  I would like to 10 solid layers.

closed 2026-09-29 by prusaslicer-bot[bot]