prusa3d/PrusaSlicer #2356: en_DE.UTF-8 crashes

Zettymaster, 2019-05-23:
### Version
version_2.0.0-37-g33f25ae98

### Operating system type + version
Arch Linux, upgraded to latest (as of 23.5.)

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
I have my user use English, but measurements, time etc. in German (easier especially for dates).
When I try to run prusa-slicer i get a std::runtime_error : 
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  locale::facet::_S_create_c_locale name not valid
Aborted (core dumped)
```

my relevant environment variables are: 
```
LANG="en_US.UTF-8"
LANGUAGE="en_US"
LC_COLLATE="en.DE_UTF-8"
LC_MEASUREMENT="en_DE.UTF-8"
LC_MONETARY="en_DE.UTF-8"
LC_NUMERIC="en_DE.UTF-8"
LC_TIME="en_DE.UTF-8"
```
but even after changing the LC_ ones to "en_US.UTF-8" i still get the same error.

_Is this a new feature request?_
not really?!

#### Project File (.3MF) where problem occurs
N/A

bubnikv, 2019-05-24:
You may need to regenerate your locale tables.
(likely a  locale-gen tool on arch-linux).

https://www.thomas-krenn.com/en/wiki/Configure_Locales_in_Ubuntu
https://unix.stackexchange.com/questions/246846/cant-generate-en-us-utf-8-locale

It is true, that the application should not crash, but give a reasonable
error report.

Zettymaster, 2019-05-24:
Trying to fall back to English  would probably be a reasonable option too, especially if said locale begins with "en". I will try your solution out when i ab home.

Zettymaster, 2019-05-24:
Update: turns out that $LC_ALL was unset, other locale settings were OK.

Zettymaster, 2019-05-24:
With all locale settings set to en_US.UTF-8 it works. But still not nice, since I like my meters, grams and prices displaying in US formats. I'm guessing it would take more then worth restructuring to make it work properly. I will open a separate Issue with a feature request for it.

closed 2019-05-24 by Zettymaster