prusa3d/PrusaSlicer #2361: Netfabb repair icon doesn't work in macOS environment

liftbag, 2019-05-24:
### Version
_PrusaSlicer-2.0.0+_

### Operating system type + version
_macOS 10.4.5_

### 3D printer brand / version + firmware version (if known)
_not relevant_

### Behavior
*  _Netfabb online repair icon in the right side bar cannot be activated with right click as in windows environment._
* ### Steps needed to reproduce the problem
  * _Load a defective stl that needs to be repaired_
  * _select the stl to be repaired with a left click_
  * _right click on the exclamation mark_
* ### Expected Results
A window should appear while the file is sent to netfabb
* ### Actual Results
  * _only the contextual menu appears_

#### Project File (.3MF) where problem occurs
[Lagarto_v5.1.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3216022/Lagarto_v5.1.3mf.zip)

bubnikv, 2019-05-24:
It will not, as this is a Windows 10 feature, sorry. We are using the
netfabb service provided by Windows 10 only.

liftbag, 2019-05-24:
> It will not, as this is a Windows 10 feature, sorry. We are using the netfabb service provided by Windows 10 only.
> […](#)

Ok, understand. Going to close.

closed 2019-05-24 by liftbag