prusa3d/PrusaSlicer #2362: machine_max_acceleration_extruding as placeholder

guestisp, 2019-05-24:
### Version
2.0.0

_Is this a new feature request?_

Yes

Currently, is not possible to set a travel acceleration different than the printing acceleration.
`machine_max_acceleration_extruding` is setting both `T` and `P` arguments to the `M204`

Also, there isn't any placeholder for `machine_max_acceleration_extruding`, I can't override the M204 with a custom start gcode, something like the following:

`M204 P{machine_max_acceleration_extruding} T1000`

I would suggest to add some additional placeholder.

I could try to do a PR on my own, if someone points me to the code generating the placeholder, I did a simple search and found nothing.

bubnikv, 2019-05-24:
Our start code contains
`M204 S[machine_max_acceleration_extruding] T[machine_max_acceleration_retracting]`
so the machine_max_acceleration_extruding should work

bubnikv, 2019-05-24:
https://github.com/prusa3d/PrusaSlicer/issues/1089 explains a lot about the M204 support

guestisp, 2019-05-24:
I'll try. I've looked in the wiki and these placeholder are not referenced.

bubnikv, 2019-05-24:
I have updated the wiki yesterday with the comment on how to get the list
of placeholders.

guestisp, 2019-05-24:
Anyway, I think that default M204 should be removed if present in the custom start gcode, like for the temperatures. Currently, two M204 are set, the "stock" one and the custom from the start gcode. That's good, because marlin would override the first one with the second one, but I think would be better to not write two times the same gcode

bubnikv, 2020-02-04:
Try to put
;machine_max_acceleration_extruding {machine_max_acceleration_extruding[0]}
into the start G-code. It works.
closing.

closed 2020-02-04 by bubnikv