prusa3d/PrusaSlicer #2363: Maintaining consistent AMF support

Spiritdude, 2019-05-24:
### Version
2.0.0.

### Operating system type + version
Linux

### 3D printer brand / version + firmware version (if known)
Custom printer with Marlin 1.1.8

### Behavior
* Pre 2.0.0 the AMF format was properly supported, although Slic3r PE already got stricter parsing the XML, whereas Slic3r was more tolerant
* with 2.0.0 the multi-material / multi-extruder settings in AMF has changed again: it no longer parses volume settings within the same mesh and enumerates tools (`T0`, `T1` etc in Gcode) automatically
* **the slicing is now depended on metadata included in the AMF** which lists PrusaSlicer settings as of `<metadata type="slic3rpe_config"></metadata>`

I do maintain various tools and CAD apps which produce AMF themselves, and I like to output valid AMF for Slic3r* and PrusaSlicer - but I can do this only when there is consistency. 

Please **do not assume** the only application to produce AMF files and using PrusaSlicer is PrusaSlicer itself, there are other applications.

### Solution
- Keep consistent default behavior among versions of PrusaSlicer, like assigning multiple extruders/tools for multiple volumes in the same mesh
- Do not rely on settings in the metadata in the AMF file only, **keep it optional** not a requirement

bubnikv, 2019-05-24:
> although Slic3r PE already got stricter parsing the XML, whereas Slic3r was more tolerant

This is interesting, because AFAIK the upstream slic3r AMF parser is based on the PrusaSlicer AMF parser.

> with 2.0.0 the multi-material / multi-extruder settings in AMF has changed again: it no longer parses volume settings within the same mesh and enumerates tools (T0, T1 etc in Gcode) automatically

Data would be helpful.

> the slicing is now depended on metadata included in the AMF which lists PrusaSlicer settings as of <metadata type="slic3rpe_config"></metadata>

This is bad, we shall fix it.

> I do maintain various tools and CAD apps which produce AMF themselves, and I like to output valid AMF for Slic3r* and PrusaSlicer - but I can do this only when there is consistency.

What exporters do you maintain? We are certainly interested in extending the exchange format or the exporters, so you could set the color or modifier type from the CAD.

In the future, we decided to not put too much support into the AMF export in regard to new PrusaSlicer features, but we should certainly still import the existing AMF files correctly. If that does not work, we shall certainly fix it.

Spiritdude, 2019-05-24:
If AMF isn't priority (which I can understand due the format openness), what alternatives are there to export multi-material/color geometries, what do you recommend?

I'm using PrusaSlicer and other slicers (Slic3r* and Cura*) only in command-line way, I currently try to support PrusaSlicer for the command-line tool https://github.com/Spiritdude/Print3r (the current version does not yet). I'm working on an unpublished JavaScript CAD framework, alike OpenJSCAD.org (which I maintained for several years) and currently try to provide consistent multi-material/color support:
- CAD: 3d objects get properties with CMY or RGB or material information
- CAD: exports AMF with different volumes using the same mesh
- Slicer: slices with different Gcode `T`*n* per volume
- Post-processing: mapping `T`*n* Gcode back to CAD model CMY and RGB coloring, e.g. using `M165` or `M163/M164` to set colors per tool with Diamond Hotend

so, I rely on the automatic assignment of tools per volume in AMF.

I attach `cube1.amf` and `cube2.amf` (metadata removed), and `prusa-slicer` on command-line slices the `cube1.amf` correctly with multiple tools, but not `cube2.amf`

[cube1.amf.txt](https://github.com/prusa3d/PrusaSlicer/files/3216552/cube1.amf.txt)
[cube2.amf.txt](https://github.com/prusa3d/PrusaSlicer/files/3216553/cube2.amf.txt)

bubnikv, 2019-05-24:
Going forward, you should generate 3MFs. The easiest for you is likely to
export a 3MF file from the GUI and then analyze it.

> https://github.com/Spiritdude/Print3r

Yes, I love it, it is written in Perl!

> CAD: 3d objects get properties with CMY or RGB or material information

In PrusaSlicer's (or Slic3r PE's) AMF or 3MF, as of now colors are assigned
either by the extruder color property or a filament color property.

> CAD: exports AMF with different volumes using the same mesh

I don't understand.

> Slicer: slices with different Gcode T*n* per volume

PrusaSlicer supports object specific presets. Just assign them in the UI
and analyze the output AMF or 3MF. We recommend 3MF going forward.

> Post-processing: mapping T*n* Gcode back to CAD model CMY and RGB
coloring, e.g. using M165 or M163/M164 to set colors per tool with Diamond
Hotend

We certainly do not support that as of now.

> so, I rely on the automatic assignment of tools per volume in AMF.

I just checked your cube1.amf and cube2.amf. Both are loaded correctly into
the GUI, including the correct extruder assignment. It is quite possible
that the command line slicer is now broken for some scenarios, as we
rewrote the whole CLI front end (well, we took most of it from the
upstream) into C++, so it is a completely new code. Good thing is, that the
command line slicer now is able to read all presets from the AMF or 3MF.

Spiritdude, 2019-05-24:
Let me clarify, the example I gave is my work flow 1) to 4), I do not expect PrusaSlicer doing this. I only need Gcode with T*n* tool changing for each material as defined in the AMF.

bubnikv, 2019-05-24:
I see when slicing from command line, cube1.amf produced two material
G-code, while cube2.amf produced a single material G-code. It works
correctly from GUI.

bubnikv, 2019-05-24:
It looks like the auto-assignment of the extruders is broken when slicing from the command line, 

> the slicing is now depended on metadata included in the AMF which lists PrusaSlicer settings as of <metadata type="slic3rpe_config"></metadata>

Are you sure? Do you have a proof?

The two AMFs you provided only differ in that the first one contains the configuration data and the other does not.

Spiritdude, 2019-05-24:
> > It looks like the auto-assignment of the extruders is broken when slicing from the command line,

Right, this is what I experienced.

> the slicing is now depended on metadata included in the AMF which lists PrusaSlicer settings as of

> Are you sure? Do you have a proof?

You just did confirm my point: 

> The two AMFs you provided only differ in that the first one contains the configuration data and the other does not.

The configuration is enclosed in the `<metadata>`, that is what I meant.

closed 2026-09-29 by prusaslicer-bot[bot]