prusa3d/PrusaSlicer #2364: prusaslicer 2.0 release version inserts filament changes automatically

bman2000, 2019-05-24:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

bman2000, 2019-05-24:
if you add layer changes to a single model with only one color / filament, slic3rpe will automatically add an unwanted  filament change into the gcode upon slicing. This should not happen unless a deliberate + is added.

fe60, 2019-05-24:
I don't know which setting caused the issue, but I faced the same problem.

PrusaSlicer
Version:   2.0.0+
Build:     PrusaSlicer-2.0.0+-201905201652

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 10.14.5 (Build 18F132)

Project/Settings resulting in .gcode containing unwanted filament change can be found in the attached HookIssueWithFilamentChange.zip (had to remove .amf to enable the upload)

[HookIssueWithFilamentChange.zip](https://github.com/prusa3d/PrusaSlicer/files/3217591/HookIssueWithFilamentChange.zip)

lukasmatena, 2019-08-09:
We could not reproduce the issue. I would guess that you inadvertently added the color change yourself and just did not realize it. First implementation of colorprint feature made this kind of error quite easy to do, current version make the color changes easier to see so everyone knows what's going on at the first sight.

I'm closing the issue. If there are more problems like this, I'll reopen it.

closed 2019-08-09 by lukasmatena