prusa3d/PrusaSlicer #2365: Make PrusaSlicer respect regional Formats

Zettymaster, 2019-05-24:
### Version
version_2.0.0-37-g33f25ae98 

### Operating system type + version
(arch) Linux

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Currently, if you have a locale set, say "en.US_UTF-8" but you want regional formats like Euros, grams and meters and date, the $LC_* environment variables get set to (in my case) "en.DE_UTF-8" and makes prusa-slicer crash with an unhelpful error message (see #2356 ). 

I propose to change this to respect the regional formats. I am willing to help if pointed in the right direction.

_Is this a new feature request?_
Yes.

#### Project File (.3MF) where problem occurs
N/A

bubnikv, 2019-08-23:
PrusaSlicer will newly provide a hint to the user if the active locale was not installed correctly in the system
Closing.

closed 2019-08-23 by bubnikv

narzul, 2020-01-24:
That is not true it just crashes.
Having the same problem that he reference and doing sudo locale-gen does not help, second changing everything in etc/locale.conf to en_us.UTF-8 does not work either.

An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid

lual, 2020-11-26:
I found a "solution" to start prusa-slizer [here](https://forum.kde.org/viewtopic.php?f=289&t=168868) 
Supply the locale as part of the start command:
`LC_ALL=en_US.utf-8 prusa-slicer`

magnificu, 2021-08-04:
Prusa slicer 2.3.3 does not start:
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid

If I set the to "LC_ALL=en_US.utf-8" it starts, but this will change my desktop locale appearance to en_US.utf-8
Is is possible to address this issue in the: Prusa slicer?

My locale settings:
LANG=en_DE.UTF-8
LANGUAGE=
LC_CTYPE="en_DE.UTF-8"
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE="en_DE.UTF-8"
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES="en_DE.UTF-8"
LC_PAPER=de_DE.UTF-8
LC_NAME=de_DE.UTF-8
LC_ADDRESS=de_DE.UTF-8
LC_TELEPHONE=de_DE.UTF-8
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION=de_DE.UTF-8
LC_ALL=

bubnikv, 2021-08-04:
What issues are you currently facing?

In PrusaSlicer 2.4.0 we will show the decimal separator according to user
locales. Currency is a difficult topic, as we don't keep a database of
material pricing in local currency. What else do you think does not
follow the user locales?

magnificu, 2021-08-04:
The issue is that if LC_ALL is not set PrusaSlicer 2.3.3 does not start.

magnificu, 2021-08-04:
I am using: "PrusaSlicer-2.3.3+linux-x64-202107161032.AppImage"

magnificu, 2021-10-18:
As a workaround I created a PrusaSlicer.desktop file with the following content:

[Desktop Entry]
Exec=LC_ALL=en_US.utf-8 "PrusaSlicer-2.3.3+linux-x64-202107161032.AppImage"

peterge1998, 2021-11-05:
@xarbit told me to report my issue with PrusaSlicer running in Flatpak here too. (Fedora 35)

The issue after installing  PrusaSlicer from Flathub:
```
➜  ~ flatpak run com.prusa3d.PrusaSlicer  
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid
```

xarbit, 2021-12-12:
Just a heads up, there will be a temporary built in workaround that addresses this issue for the upcoming 2.4.0 release, only for the flatpak build on flathub.

https://github.com/flathub/com.prusa3d.PrusaSlicer/blob/beta/entrypoint

Bricksone, 2022-08-11:
Will this ever be fixed? It's still having the same problem in 2.4.2-7 (arch repo), 2.6.0 (AUR) or 2.4.2 (flathub).

xarbit, 2022-08-11:
@Bricksone probably not as much as I understood .. my workaround for flathub was the only solution to this. Ill look in to see if I can find what is going on.

MightyPork, 2023-03-03:
prusaslicer worked fine, but since some recent update (possibly update of something else in the system), opening 3mf files from a file manager no longer works and I see the above mentioned locale error in system log. However, if I run PrusaSlicer manually, it works OK.

I also use "per partes" locale, because my system is in English, but I want local date/time and units.

Editing the desktop file like this works, hopefully with no adverse effects, but it's a hack I'm not too happy with.

[image]

chiboreache, 2023-08-22:
here is how that string looks in `.destop` file in KDE Plasma

```sh
Exec=env 'LC_ALL=en_US.utf-8' prusa-slicer %U
```

barnabywalters, 2023-09-07:
I have the same issue in Kubuntu with Prusa Slicer 2.6.1. Downloading, extracting and running the AppImage has no externally visible effect. Running it from a terminal shows the “An error occured while setting up locale.” error message. Running the two commands it suggests fails on my system. Starting Prusa Slicer manually from the terminal while specifying LC_ALL=en_GB.utf8 does work.

This issue makes the Prusa Slicer AppImage appear completely broken to most users on all systems which are effected by it.

Mojitas, 2024-07-13:
I just installed manjaro and encountered this exact problem. 
But setting LC_ALL=sv_SE.utf-8 worked for me. It does not seem to matter which locale is set.

SamsiFPV, 2024-12-20:
I have the same issue, Fedora KDE. launching it with `flatpak run /path-to-prusa-flatpak-folder` works though. Only the desktop file seems broken. The fix from migthypork worked, adding the environment variable to the desktop file.

S-h-i-n-r-a, 2025-04-09:
Can confirm this is still an issue as of v2.9.1.
Can also confirm the solution of setting LC_ALL to a locale of your choice either via command line or application settings is still an effective workaround.

zettymaster, 2026-09-02:
WTF? this is still a problem as of quite a new PS version (as seen by @Futrax above, just 3 weeks ago).