prusa3d/PrusaSlicer #2367: Fail to update Firmware

DuncanEastland, 2019-05-24:
### Version
2.0.0+Win64
### Operating system type + version
Windows 10
### 3D printer brand / version + firmware version (if known)
Prusa3d Mk2
### Behavior
Upgraded hardware to Mk2.5S MMU2S. Tried to flash the firmware following the instructions but flash failed. the log contained the following. The screen is also now blank on the printer.

\prusa3d_fw_3_7_1_MK3S_1_0_5_MMU2S(1)\prusa3d_fw_MK3S_3_7_1.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on May 20 2019 at 18:52:49
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM3
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\Users\duncane\Downloads\prusa3d_fw_3_7_1_MK3S_1_0_5_MMU2S(1)\prusa3d_fw_MK3S_3_7_1.hex"
avrdude-slic3r: writing flash (249438 bytes):
avrdude-slic3r: stk500v2_command(): command failed

Writing | ################################################## | 100% 43.93s

avrdude-slic3r: 249438 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\duncane\Downloads\prusa3d_fw_3_7_1_MK3S_1_0_5_MMU2S(1)\prusa3d_fw_MK3S_3_7_1.hex:
avrdude-slic3r: load data flash data from input file C:\Users\duncane\Downloads\prusa3d_fw_3_7_1_MK3S_1_0_5_MMU2S(1)\prusa3d_fw_MK3S_3_7_1.hex:
avrdude-slic3r: input file C:\Users\duncane\Downloads\prusa3d_fw_3_7_1_MK3S_1_0_5_MMU2S(1)\prusa3d_fw_MK3S_3_7_1.hex contains 249438 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 31.96s

avrdude-slic3r: verifying ...
avrdude-slic3r: verification error, first mismatch at byte 0x3c000
                0x01 != 0x51
avrdude-slic3r: verification error; content mismatch

avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude-slic3r done.  Thank you.

#### Project File (.3MF) where problem occurs

bubnikv, 2019-06-03:
Is it reproducible all the time? Do you use some low quality USB cable or a
long USB cable?

DuncanEastland, 2019-06-03:
I've tried a couple of usb cables, both less than 1m, but always the same
result.

bubnikv, 2019-06-03:
Do you have more computers? Are you able to flash your printer with another
one?

DuncanEastland, 2019-06-03:
I will try a couple more computers and a couple more usb cables

DuncanEastland, 2019-06-03:
I have tried three computers and three cables all with the same result.

If I flash firmware for MK2 it succeeds and I can see the menu on the
printer screen.

If I flash firmware for MK2.5 it says it succeeds but the printer
screenjyst shows two rows of squares and the printer emits a high pitched
whistle.

If I flash firmware for MK2.5S it says it succeeds but screen agayshows two
rows of squares, but no whistle noise.

If I flash firmware for MK3 mmu2 it says it fails.

Duncan

bubnikv, 2019-06-03:
> If I flash firmware for MK2 it succeeds and I can see the menu on the
printer screen.

You indicated that you are flashing an MK2 printer, so it is great that the firmware flashing succeeded.

> If I flash firmware for MK2.5 it says it succeeds but the printer
screenjyst shows two rows of squares and the printer emits a high pitched
whistle.

Why are you flashing the MK2.5 firmware into the MK2 printer?

> If I flash firmware for MK3 mmu2 it says it fails.

Because the MK3 control board is different.

DuncanEastland, 2019-06-03:
I had a MK2 printer but have fitted the upgrade kits to MK2.5s and mmu2
hence trying to flash with the upgraded firmware.

rtyr, 2021-01-29:
I believe this was already resolved. Closing.

closed 2021-01-29 by rtyr