prusa3d/PrusaSlicer #2369: no access to linux file system except for /usr

jbernardis, 2019-05-24:
### Version
2.0.0+linux64

### Operating system type + version
ubuntu 18LTS

### 3D printer brand / version + firmware version (if known)
Self built - using marlin firmware 

### Behavior
When I try to load a model into Prusaslicer, I specify "file system", but the only directory presented to me is /usr.  This is wrong.  It is completely legitimate for a user to be accessing several different directories off of root, including /tmp, /opt, /media, /var.  And any place that they shouldn't be writing to they most likely do not have write privileges.  In my case, I have a Network Attached Storage mounted to directory /netgear.  I do plenty of design work on other machines in my house, and use the NAS as a common point to place stl and gcode files.

This is not an appimage limitation because my cura and pathio appimages allow access.

#### Project File (.3MF) where problem occurs
It's not project-specific

guestisp, 2019-05-24:
I don't have this behaviour, and i'm using your exact configuration (Ubuntu 18 64bit)

jbernardis, 2019-05-24:
I just verified it.  Whether I hit the + button on the top of the screen, use import from the menu, or open project from the main menu, when I choose "file system" all I see is /usr.  In fact when I try to export anything, including gcode, I have the same limitation.  I even went into search and tried to type in the full path, but it didn't accept what I was typing - it just did nothing.

I also just went again into cura, slic3r, and pathio - all of which are appimages - and they work just fine.  I can see the entire disk.  This behavior only seems to come from prusaslicer; it's a shame because I really like it.

jbernardis, 2019-05-24:
SOLVED.  I had overlooked the fact that there is a "pencil" button on the upper left that allows me to enter any path. 

The list of locations still has a "file system" that continues to only show me "/usr", but use of the pencil button if fine by me.

closed 2019-05-24 by jbernardis

bubnikv, 2019-06-03:
This is how the dialog looks on Ubuntu 16.

[image]

At least it gives me access to the home directory. I have no idea why entering a single slash to the top line does not show the full content of the root. Maybe it is some security measure of the GTK2?

thekarinka, 2022-11-14:
See this https://github.com/ivocavalcante/PrusaSlicer-snap, it's need to connect removable media for snap