prusa3d/PrusaSlicer #2370: Can not import config bundle

AndyWo83, 2019-05-24:
### Version
Version: 2.0.0+win64
Build: PrusaSlicer-2.0.0+win64-201905201652
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 7 (build 7601, Service Pack 1), 64-bit edition

### Behavior
I have saved my config bundle for a friend.
He can not import it, because he become an error message.
[image]

bubnikv, 2019-06-03:
Please attach your config bundle from your friend.

Migamix, 2019-06-15:
does the slicer crash and close when you save the bundle?. 
it seems to stop saving the file when it gets to the SLA presets all of the time. the file is truncated and incomplete. 
last line on every bundle i export
`[presets]`
`print = 0.15mm QUALITY MK3`
`sla_prin`

bubnikv, 2019-06-17:
it is a known issue, it will be fixed in the next release. Duplicate of #2307, #2336.

wildblue, 2019-07-13:
open the file  PrusaSlicer.ini in your config-folder
maybe also check the file  slic3r.ini

change
<
sla_material = 
sla_print = 
>
into
<
sla_material = Prusa Orange Tough 0.05
sla_print = 0.05 Normal
>
open prusa slicer and export your config bundle.

background:
the file vendor/PrusaResearcch.ini is invalid
following must be changed to work on initializing
<
[sla_print:*common*]
compatible_printers_condition = printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and
>
into
<
[sla_print:*common*]
printer_technology = SLA
compatible_printers_condition = 
>

and
<
[printer:Original Prusa SL1]
printer_technology = SLA
...
>
into
<
[printer:Original Prusa SL1]
inherits = *common*
>
this change results in the correct behaviour, but only if not run once without this fix.
the old config generates the empty config-valurs and these can only be fixed manually at the moment.

hey prusa-team, what is taking so long explaining this or releasing a bugfix? almost2 month since the last release.

don't you have enough developers? i would work for your company, having 10 years experience in c++, c#, digital image recognition, electronics, mechanics. have built my own 3d printer not long ago completely from scratch.

maybe i hear from you, i am located in germany, but could move when nessesary.

with the best regards in this short time

ben

lukasmatena, 2019-07-13:
@wildblue I already replied in a different issue (https://github.com/prusa3d/PrusaSlicer/issues/2446#issuecomment-511150892). Please, do not duplicate posts.

bubnikv, 2019-08-08:
This issue will be fixed by the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-08 by bubnikv