prusa3d/PrusaSlicer #2372: Can't upload .3mf files to GitHub for issues

tobymurray, 2019-05-24:
Attempting to follow this step of the issue template:

> #### Project File (.3MF) where problem occurs
> _Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

yields (for me):
[image]

I imagine it's worthwhile to have the instructions work without any fuss, so if there's a recommended workflow for attaching `.3mf` to GitHub issues I think it's worthwhile to document in the issue template.

termlimit, 2019-05-24:
You can zip the 3mf. Is that an option?

tobymurray, 2019-05-24:
Definitely that's an option. In the linked issue I just added a `.txt` on the end and that seemed to work fine. 

I just wanted to raise that if that's what needs to be done the issue template should reflect that. It seems prudent to lower the friction for users to upload the required context if they've gone to the effort of reproducing an issue.

bubnikv, 2019-06-03:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/050e312037720920b149780df4998c053248e864

closed 2019-06-03 by bubnikv