prusa3d/PrusaSlicer #2374: Can't import settings from gcode produced from slic3r and Prusa Slicer itself.

Neofitos, 2019-05-24:
2.0.0

Windows 10

MK2.5 FW 3.7.1

Can't import settings from gcode produced from slic3r and Prusa Slicer itself.

[image]

eowindel, 2019-05-25:
Same issue for me !

dolphe, 2019-05-26:
Confirmed
Windows 10
MK3s FW 3.7.1

mascalx, 2019-05-29:
Same issue. Changing the first line of the gcode from:

; generated by PrusaSlicer...

to:

; generated by Slic3r Prusa Edition...

solves the issue. When programmers changed the name of the program to PrusaSlicer they probably just forgot to update the import section.

Neofitos, 2019-05-29:
@mascalx is not working for me. As I wrote, I'm getting the same error even if I try to open file that is generated from the PrusaSlicer 2.0.0 itself.

mascalx, 2019-05-30:
@Neofitos that's a strange thing. I can import the configuration from other gcodes even by just changing PrusaSlicer to Slic3er in the first line of the gcode file using a text editor. I tried with several files and it worked every time...

FidelCapo, 2019-05-30:
Thank you for report.

Created JIRA-955.

closed 2019-05-30 by FidelCapo