prusa3d/PrusaSlicer #2375: Icons too big

M4rc3lv, 2019-05-25:
### Version
Version 2.0.0-Linux64

### Operating system type + version
Linux Mint 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
On my screen the icons are so big that I can not click on the button to add a STL file. Other applications I run don't have this, only Slic3r.
[image]

tkircher, 2019-05-25:
Try making the window bigger.

M4rc3lv, 2019-05-25:
My window was maximized. My solution now is to use an older version.

iconoclast, 2019-05-25:
[image]

As the screen shot illustrates, this bug is new with the release of "PrusaSlicer" 2.0.  
It wasn't a problem with Slic3r-PE even through 1.42-beta2.

JohnOCFII, 2019-05-26:
@M4rc3lv - It would be more useful for the developers if you could add specific information about your system.  What version of Mint?  What graphics card in your computer, etc.

M4rc3lv, 2019-05-26:
Linux Mint 19.1 Cinnamon
Cinnamon version 4.0.10
Linux kernel 4.15.0.50 generic
Graphics card Intel HD Graphics 530

YuSanka, 2019-05-27:
Close as a duplicate of #2247.

closed 2019-05-27 by YuSanka