prusa3d/PrusaSlicer #2376: Problem with support generation

thierry7100, 2019-05-25:
### Version
2.0.0 linux64 but also present in 1.41.3 

_Use `About->About PrusaSlicer` for release versions_
### Operating system type + version
Ubuntu 19.04

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3

### Behavior
With some stl files, Slic"r generates may be too much supports, so there is no filament for object itself.
See the attached images. 
First one is the object to be printed.
Second one is after slicing, the axe's handle is not present ! So of course when it try to remove support, the axe is not anymore attached !

#### Project File (.3MF) where problem occurs
Here is my files (first screen shots)
[image]
[image]

then the .3MF file (ZIP because not able to upload .3MF directly ?)
[Uploading Chibi Poh corps noir.zip…]()

bubnikv, 2019-06-03:
Just looking at your screesnhot, the axe is not a valid object 
https://en.wikipedia.org/wiki/Manifold

There is an exclamation mark in the "manipulation de l'Object" box. If you hover the mouse over the exclamation mark, you will get information on the model errors. 

you may try https://service.netfabb.com/login.php to fix the model before printing

thierry7100, 2019-06-03:
Hi

Thanks, the axe handle is indeed bogus, artère à repaire with netfabb printing in now fine.

bubnikv, 2019-06-03:
closing then.

closed 2019-06-03 by bubnikv