prusa3d/PrusaSlicer #2378: Wipe tower is included in auto placement

jsilva-iii, 2019-05-25:
### Version
2.0.0+win64

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Mk 2.5S+MMU2S FW 3.7.1

### Behavior
When press the arrange Icon, wipe tower is part of placed objects i- not usual behaviour. Skirt generation then is through the wipe tower

[image]

[5xtest cubes.zip](https://github.com/prusa3d/PrusaSlicer/files/3219521/5xtest.cubes.zip)

lukasmatena, 2019-05-27:
Thanks for reporting. There is an open pull request https://github.com/prusa3d/PrusaSlicer/pull/2198 from @tcm0116 that contains a fix for this. Hopefully we'll find time to review and integrate it soon.

Patronics, 2019-07-23:
I experienced the same issue on macOS, was about to create another issue when I found this one. Hopefully that pull request is merged soon.

lukasmatena, 2019-10-08:
Fixed by @tcm0116 in https://github.com/prusa3d/PrusaSlicer/pull/2198.

[image]

The fix is in master now and will be part of PrusaSlicer 2.2.0. Closing the issue.

closed 2019-10-08 by lukasmatena