prusa3d/PrusaSlicer #2381: Gap in sliced model

Nicornru, 2019-05-25:
### Version
_2.0.0 - win64_
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Win10 x64
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### Behavior
* _Describe the problem_
Gap in sliced model
* _Steps needed to reproduce the problem_
Just slice the model
* _Expected Results_
No gap in model
* _Actual Results_
[image]
[image]
Same file in Cura 4.0:
[image]

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
STL: [080_schwein_full.zip](https://github.com/prusa3d/PrusaSlicer/files/3219547/080_schwein_full.zip)
3mf: [3mf_080_schwein_full.zip](https://github.com/prusa3d/PrusaSlicer/files/3219550/3mf_080_schwein_full.zip)

Sebastianv650, 2019-05-25:
This part of the tail is too thin to be printed with your selected extrusion width. Cura has an inovative approach to work around this, that's why it's working there - how good this works is depending on the situation.

For devs, if you want to have a look at this feature: It reduces the flow on the "way back" to achieve a thinner wall than normaly possible, using the already existing perimeter as a one side limitation for the flow. This doesn't look perfect all times, but it seems to work surprisingly well.

BerndJM, 2019-05-25:
Unchecking "detect thin walls" should solve this ...

Nicornru, 2019-05-26:
> 
> 
> Unchecking "detect thin walls" should solve this ...

Tried, no effect.

ilium007, 2019-06-05:
I am getting gaps in walls as well. I thought checking "Detect thin walls" would fix this.

As you can wee in the attached image other walls are being printed perfectly - I am getting a wall gap around the small ring.

[image]

rtyr, 2020-04-17:
It is OK with "detect thiw walls" disabled.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]