prusa3d/PrusaSlicer #2382: NVIDIA OpenGL driver crashes when viewing the Platter after adding a printer (insufficient GPU memory).

spot1984, 2019-05-25:
### Version
PrusaSlicer 2.0.0 +Win64
Downloaded installer, not zip file.
(Suggestion: It would be nice to be able to copy that to the clipboard from the about dialog)

### Operating system type + version
Windows 10 x64 Home

System summary:

> OS Name	Microsoft Windows 10 Home
> Version	10.0.17134 Build 17134
> Other OS Description 	Not Available
> OS Manufacturer	Microsoft Corporation
> System Name	_[redacted]_
> System Manufacturer	Gigabyte Technology Co., Ltd.
> System Model	GA-770T-USB3
> System Type	x64-based PC
> System SKU	
> Processor	AMD Athlon(tm) II X3 445 Processor, 3100 Mhz, 3 Core(s), 3 Logical Processor(s)
> BIOS Version/Date	Award Software International, Inc. F4, 8/26/2010
> SMBIOS Version	2.4
> Embedded Controller Version	255.255
> BIOS Mode	Legacy
> BaseBoard Manufacturer	Gigabyte Technology Co., Ltd.
> BaseBoard Model	Not Available
> BaseBoard Name	Base Board
> Platform Role	Desktop
> Secure Boot State	Unsupported
> PCR7 Configuration	Binding Not Possible
> Windows Directory	_[redacted]_
> System Directory	_[redacted]_
> Boot Device	_[redacted]_
> Locale	United States
> Hardware Abstraction Layer	Version = "10.0.17134.765"
> User Name	_[redacted]_
> Time Zone	Central Daylight Time
> Installed Physical Memory (RAM)	10.0 GB
> Total Physical Memory	10.0 GB
> Available Physical Memory	4.50 GB
> Total Virtual Memory	20.0 GB
> Available Virtual Memory	11.1 GB
> Page File Space	10.0 GB
> Page File	_[redacted]_
> Kernel DMA Protection	Off
> Virtualization-based security	Not enabled
> Device Encryption Support	Reasons for failed automatic device encryption: TPM is not usable, PCR7 binding is not supported, Hardware Security Test Interface failed and device is not InstantGo, Un-allowed DMA capable bus/device(s) detected, TPM is not usable
> Hyper-V - VM Monitor Mode Extensions	Yes
> Hyper-V - Second Level Address Translation Extensions	Yes
> Hyper-V - Virtualization Enabled in Firmware	Yes
> Hyper-V - Data Execution Protection	Yes

PrusaSlicer System Information

> PrusaSlicer
> Version:   2.0.0+win64
> Build:     PrusaSlicer-2.0.0+win64-201905201652
> 
> Operating System:    Windows
> System Architecture: 64 bit
> Windows Version:     Windows 10 (build 17134), 64-bit edition
> 
> OpenGL installation
> GL version:   3.3.0
> Vendor:       NVIDIA Corporation
> Renderer:     GeForce 9800 GTX/9800 GTX+/PCIe/SSE2
> GLSL version: 3.30 NVIDIA via Cg compiler
> Installed extensions:
> 
> GL_ARB_ES2_compatibility
[... 243 lines cut ...]
NVIDIA OpenGL driver crashes when viewing the Platter after adding a printer.

Detail:
Downloaded and installed PrusaSlicer 2.0.0+win64
The first time I ran it after the wizard, and every time I re-run the software... 
When I add the **Prusa i3 MK2s** model I get an **NVIDIA OpenGL Driver** error dialog box:
> An application has requested more GPU memory than is available in the system.  
> The application will now be closed.
>
> Error code: 6
> Would you like to visit
> http://nvidia.custhelp.com/app (bla bla bla I don't want to type the whole url)
and the PrusaSlicer software exits.
Subsequently, when I run the software I can see the 3D **Platter** view and adjust the 3D view of the generic buildplate area using the mouse so 3D rendering is working.  There are no printers installed.  If I install the printer from a view that does not show the platter, it appears to work and I can look at the **Printer**, **Filament**, or **Print** views, but as soon as I return to the **Platter** view, there is a delay, the screen does not change (still shows the previous view) and I get the NVIDIA OpenGL error.
Also tested adding the **Prusa i3 MK3** printer with same results.

I've included my system information.

This PC is not the latest and greatest but I it for CAD, 3D printing, and light gaming.  I doubt it is unable to handle PrusaSlicer.

I really can't wait to use PrusaSlicer, it looks FANTASTIC.
Nice work and good luck!

#### Project File (.3MF) where problem occurs
Not applicable

enricoturri1966, 2019-05-27:
@spot1984
We are already trying to let PrusaSlicer be less hungry in regard to GPU memory (http://dev.prusa3d.com/browse/SPE-953)
Could you please try if this experimental build runs on your pc ?

https://www.dropbox.com/s/yldd5pwq1w8pqan/PrusaSlicer-2.0.0%2B59-et-experiments-win64-ge0da08906-201905270722.zip?dl=0

MartyZzzz, 2019-05-31:
I have exactly the same issue with low GPU on my notebook (HP Elitebook 8730w, Win10 32bit) after installing Prusa Slicer ver.2. 
In previous Slicer versions no problem with GPU.

Delelan, 2019-06-10:
Tried the experimental version that was linked above and I can now add a printer profile to slicer. The OpenGL driver has stopped crashing.

scottmi01, 2019-06-11:
Same problem with Nvida GeForce 210 board running on windows 10 64-bit and i3mk3. Copied files from the above experimental build (PrusaSlicer-2.0.0+59-et-experiments-win64-ge0da08906-201905270722) over the installed files. This worked to run, import an STL, export a gcode file, and save and load a project... so far, so good.

bubnikv, 2019-06-14:
thanks for the report.

bubnikv, 2019-08-01:
The upcoming PrusaSlicer 2.1.0-alpha1 implements the texture compression above, it should solve these issues. Closing.

closed 2019-08-01 by bubnikv