prusa3d/PrusaSlicer #2383: Slow Processing/Low Ultilization

CAP1Sup, 2019-05-25:
Version: 1.42.0 Alpha 7 (Although this probably applies to the new versions as well)

Platform: Windows 10, on a laptop with an 8th gen i5

Printer: Anet A8, but not really a consideration

Problem: 
Slic3r is takes a long time on larger parts and a quick look at task manager says that the CPU is only being used 20-30% and GPU (Integrated) isn't being used at all. Is there any way that Slic3r could be modified to support using more power for faster computations?

bubnikv, 2019-06-03:
Most of the code is highly parallel. What is not parallel is the G-code
export including the slow "avoid perimeters" feature.

gysnst2, 2019-06-03:
I’ve noticed on larger slices it can take forever to slice.  If I scale the model down 20-40 percent it slices wicked quick!

closed 2021-02-23 by CAP1Sup