prusa3d/PrusaSlicer #2384: Viewing firmware update log hangs Slicer

DaveGadgeteer, 2019-05-25:
### Version
_Version of PrusaSlicer used goes here_
2.0.0+
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
MacOS 10.14.4
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa mk3
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Configuration->update firmware
Whether the update succeeds or not (first tries my usb connector wasn't pushed in hard enough on Mac, so it failed) when you click on View Log Slicer goes into an unresponsive state.
I've seen this same behavior before, with previous Slic3r.
_Is this a new feature request?_
no
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

vojtechkral, 2019-05-28:
Can you confirm / describe the steps needed to reproduce the problem? I assume you select a firmare hex file and press `Flash!`, then try to open the log. Is there something special that needs to be done to make the problem happen? Is there something special about your system? Thank you.

DaveGadgeteer, 2019-05-28:
There isn't anything special that I know of. 
I'm using a current Mac book Pro, which has only USB C connectors, so I connect through a C to 3 adapter.
The first 2 attempts failed because the adapter connection to C wasn't pushed in far enough. I tried to read the log then, but Slicer hung and had to be force quit. The next time I discovered the connection wasn't right, based on YouTube IIRC, I learned it should show my printer info and discovered the unseated connector. Then the firmware updated successfully, and I tried to read the log just to see if it ever works. Months ago I had your slic3r again hang on trying to read the log, so it's not a new problem with the new Slicer, but a long-standing one.
I have never managed to view the log since I bought the mk3.
Dave

rtyr, 2021-01-29:
Should be fixed. Closing.

closed 2021-01-29 by rtyr