prusa3d/PrusaSlicer #2387: Printing blocked on firmware update available message

brjhaverkamp, 2019-05-26:
### Version
Version 2.0.0 linux 64
### Operating system type + version
Ubuntu 19.04

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
I have set up prusaprinter with octoprint.
When sending a job to the printer, the print is not started, but on the display of the printer a notification about new firmware is shown. Only after I pres the knob, the print is started.

The desired behaviour is that the notification of new firmware is shown in prusaslicer and the option is given there to update the firmware. (preferably via octoprint.)

bubnikv, 2019-06-03:
I have a good news for you, that the firmware team is working on time-outing this message.
Closing here at the slicer issue tracker.

closed 2019-06-03 by bubnikv

MucaroBoricua, 2019-12-30:
That was back in June.  Today is December 30 and this hasn't been fixed yet.  : (