prusa3d/PrusaSlicer #2389: Generated GCODES do not have commands to Hotend heating

Xeliaa, 2019-05-26:
### Version
PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
Ender 5 / Klipper Firmware + Octoprint

### Behavior
Will not heat up the hot end after heating up the bed, have to reslice model many
times in order for it to get the commands to heat up the hotend

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
[nana2_1.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3220363/nana2_1.3mf.zip)

bobstro, 2019-06-01:
Several people using non-Prusa printers have posted on the forums with similar issues. In at least one case, they were using custom startup gcode that differed significantly from the Prusa-provided start gcode. The custom gcode did not include any M104 commands, and it appears PS was inserting them _after_ the custom gcode, including after the prime line print, which resulted in failures. If PS is detecting the absence of warmup commands and auto-inserting them if missing, it should probably happen before custom gcode is invoked.

bubnikv, 2019-08-01:
> If PS is detecting the absence of warmup commands and auto-inserting them if missing, it should probably happen before custom gcode is invoked.

Maybe it should not insert these commands at all. This is a never ending story, and everyone will have its opinion whether the command should be inserted before or after.

lukasmatena, 2023-01-21:
Duplicate of #2420.

closed 2023-01-21 by lukasmatena