prusa3d/PrusaSlicer #2392: Unknown unhandled error slicing a simple object

marclefevre, 2019-05-26:
### Operating system type + version
PrusaSlicer
Version:   2.0.0+win64
Build:     PrusaSlicer-2.0.0+win64-201905201652

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17134), 64-bit edition

OpenGL installation
GL version:   4.6.0 NVIDIA 397.31
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1050/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
[... 308 lines cut ...]
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Homemade 3D printer running Marlin 1.1.9

### Behavior
"Slice Now" causes "Unhandled unknown exception"
Load the file and slice

#### Project File (.3MF) where problem occurs
I tried to upload my 3mf file but github says that that file type is not supported.  I zipped the 3mf file and that worked.
[Filament Guide Rod Support.zip](https://github.com/prusa3d/PrusaSlicer/files/3220579/Filament.Guide.Rod.Support.zip)

marclefevre, 2019-05-26:
A follow-up error occurs that says "Internal error; bad allocation" and the object on the plater has horizontal breaks in the sliced version.

marclefevre, 2019-05-26:
[unhandled error bad allocation.txt](https://github.com/prusa3d/PrusaSlicer/files/3220582/unhandled.error.bad.allocation.txt)

marclefevre, 2019-05-26:
I tried slicing other objects and they all fail the same way now in Prusa Slicer.  I tried the original object in Slic3r PE and it slices without error.

BirkBinnard, 2019-05-26:
FWIW I sliced your part ok with both PrusaSLicer and Craftware. It's a pretty simple part - nothing obviously tricky about it. Have you tried removing and reinstalling PrusaSlicer?

marclefevre, 2019-05-26:
Thanks.  I was holding off on reinstalling it until I knew there was no
more data to collect from the current installation.  The current install
was working fine on all parts until today and now it won't slice any
parts.  Should I try reinstalling now?  Does it matter if I uninstall first?

BirkBinnard, 2019-05-26:
Always uninstall first.

marclefevre, 2019-05-27:
Reinstalling PrusaSLicer fixed the problem.  Since we didn't actually find the problem, I'm not sure it makes sense to close the issue but I don't see anyone working on it at this point so I will.

closed 2019-05-27 by marclefevre