prusa3d/PrusaSlicer #2393: Filament Swap (M600) added to wrong layer

numanair, 2019-05-26:
### Version
2.0.0 win64

### Operating system type + version
Windows 10 1903

### 3D printer brand / version + firmware version (if known)
Single extruder

### Behavior
**Intended behavior:**
When clicking the + button to add a material change, the change should happen at the end of the currently selected/displayed layer.

**Actual behavior:**
After clicking the + button an M600 is added two layers later. This is reflected in the exported G-Code as well as the "color print" 3D preview, but not always the layer slider on the right of the UI. 

Next I delete the erroneous color swap and attempt to add it at the correct layer. This causes the layer change to be inserted a layer before the intended point (at least as I understand the UI here).

Additionally, clicking "Slice Now" at this point clears the color change from the layer slider, but not entirely from the 3D preview. Adding another swap seems to force an update of the slider UI and reveals both swap points.

If more specific repro steps are required let me know.

#### Project File (.3MF) where problem occurs

[example_swap.zip](https://github.com/prusa3d/PrusaSlicer/files/3221058/example_swap.zip)

wabbitguy, 2019-05-27:
I use the filament change on my STL files, not .3MF so perhaps the problem pertains to 3MF files.

For STL when I use the (+) function I found it was programmed for the filament change using the same logic as their colour change on PRUSA's web site.

https://blog.prusaprinters.org/color-print/

Select the layer for the change, and that's where it puts the M600. So the filament change is inserted when I do it, at the beginning of the layer I currently have selected and that seems the logical way to do it to me.

FidelCapo, 2019-06-03:
@numanair I'm sorry, I didn't understand your problem. Can you please make short video or describe your issue on screenshots? It would be very helpful. Thank you

YuSanka, 2019-06-20:
Thanks for your report. We can to reproduce it and its fixing is in progress [SPE-959](https://dev.prusa3d.com/browse/SPE-959)

CJx83, 2019-08-15:
Also had this issue. The color bars to the right side are confusing, and it is not clear at which layer the color change will be inserted. I made this video today to communicate the issue to Joel Telling and thank him for some inspiration for a cool project I did with a class I am teaching

https://www.youtube.com/watch?v=GU2cavlImFE

YuSanka, 2019-08-16:
Actually, no :slightly_smiling_face:. Initial issue was fixed with 1986005 by @bubnikv, but probably we just forgot to close it.
What you considered as a bug, is actually the correct behavior. Every slider tick(position) corresponds to a printing layer on a 3D scene. Clicking on "+" is used for a color change on a selected layer.

CJx83, 2019-08-16:
It almost makes sense if you scroll the layers down from the top as you view them, but if you scroll up from the bottom (as the print will be built by the printer), it appears that the colors are shifted up one layer. I believe the issue could be solved by having a 0.00mm slot at the bottom and shifting all of the colors down a layer on the slider at the right side. 

I explained the issue in the video at the YouTube link in my last comment. If you are visualizing the print being built from the build plate up, it is not intuitive where to add the color change (M600) because of the colors on the slider being "off" by one layer.

bubnikv, 2019-08-19:
@CJx83 With all respect, there are many ways to skin the cat. The rule is simple and understandable: The selected layer will be the first one printed with the new color. I would say that one half of the slicer users are happy with the current behavior while the other half is unhappy.

I would recommend to you to use the up cursor key to shift one layer up if you don't like the current behavior.

closed 2019-08-23 by YuSanka