prusa3d/PrusaSlicer #2394: Missing Pieces of a print - after repair by both Netfabb and Slicer

Wot-Dabny, 2019-05-26:
### Version
Slicer version 1.41.2+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
I3 mark3 - no modifications

### Behavior
When I slice the model it is missing it's top 80%. I have tried repairing it with both Netfabb and the Slicer repair. I can not see any obvious defects in the model. I downloaded it from Thingverse.

[image]
[image]

#### Project File (.3MF) where problem occurs
[Uploading CE3_Kroxigor1_fixed_0.15mm_PLA_MK3.zip…]()

Sebastianv650, 2019-05-27:
Your uploaded .3mf file link is only a link to this issue, so it's not working. The stl itself would be also useful.

lukasmatena, 2019-05-27:
@Wot-Dabny 
1. You are using Slic3rPE 1.41.2. Many bugs have been fixed in the new PrusaSlicer 2.0.0, so it is worth to test in that version as well. 
2. As already pointed out, the link to the 3MF does not link to the 3MF.
3. Model Kroxigor1 downloaded from https://www.thingiverse.com/thing:3341531 slices fine for me in both 2.0.0 and 1.41.3.
4. Position of the layer slider in your screenshot suggests you only limit the visibility to about 20 % of the actual print height.

Wot-Dabny, 2019-05-27:
Thank You! I think I accidentallyh moved the layer slider and didn't realize it. I will install the new 2.0 as well.I really appreciate the fast informative help.
Somewhere between physics and philosophy is the intersection of the real world. 

 
 
@Wot-Dabny

closed 2019-06-03 by lukasmatena