prusa3d/PrusaSlicer #2395: Issue with Multi Part co-ordinates on 3mf files

jmone1, 2019-05-26:
### Version: 2.0.0+win64
### Operating system type + version: Windows 10 64 Bit
### 3D printer brand / version + firmware version (if known): Prusa Mk3s+MMU2s (3.7.1 / 1.0.5)
### Behavior
There is an issue where using 3mf files imported into Prusa Slicer is getting the co-ordinates of the parts wrong.  This is an issue where there are multiple parts in the 3mf file as PrusaSlicer does not keep the relative positioning of these parts (see pic and sample 3mf file).  Using this 3mf file:
1) In 3D Builder the object has co-ords of 100 x 100 x 0 - but in Prusa Slicer it seems to be positioned at 47 x 31 x 0 (ish)
2) If I have a multi part object (say just cut in half), the relative positioning of the pieces are not kept (eg both parts of the split pyramid are now overlapping).  If you have a 2 parts vertically stacked, when you open in Prusa Slicer, both parts will be on the build table

It is easy to replicate, create an object in 3D Builder, spit it.... (or create multiple objects) and save in the default 3mf.  Drag the 3mf file t into Prusa Slicer = all the co-ords are out

Thread of the Prusa Frorum: https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-hardware-firmware-and-software-help/prusa-slicer-issue-with-multi-part-co-ordinates-on-3mf-files/ 

#### Project File (.3MF) where problem occurs
- Attached PIC of the issue
- Attached 3mf file
[test multi slice colour.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/3221137/test.multi.slice.colour.3mf.txt)
[image]

enricoturri1966, 2019-05-27:
Working on it -> http://dev.prusa3d.com/browse/SPE-954

jmone1, 2019-05-27:
Thanks!  FYI - I don't have access to JIRA so I can't open that link

bubnikv, 2019-08-01:
Implemented with fab363493140a93c1786c7898424a73dd1ed2a2e
Closing.

closed 2019-08-01 by bubnikv