prusa3d/PrusaSlicer #2398: PrusaSlicer2.0.0 appimage won't open.

TheMrCaveman, 2019-05-27:
### Version
_2.0.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_linux

### 3D printer brand / version + firmware version (if known)
_MK2.5S

### Behavior
* _AppImage won't open no matter what permissions i give it.

BerndJM, 2019-05-27:
A bit more info about your operating would be helpful not only just "_linux"!
Have you tried to start the AppImage in a terminal, may be there are some error messages?!?

vojtechkral, 2019-05-28:
Yes, running the AppImage from a terminal and copying the output here would be useful.

josbouten, 2019-05-30:
I'm running ubuntu 18.04.2 LTS and XFCE4 and nvidia drivers for a GeForce GTX 1050 using Xorg
I am encountering probably the same error. I'm trying to run PrusaSlicer-2.0.0+linux64-201905201652.AppImage and the program returns this output: 

`./PrusaSlicer-2.0.0+linux64-201905201652.AppImage --loglevel 10
Gtk-Message: 11:10:47.450: Failed to load module "overlay-scrollbar"
[2019-05-30 11:10:47.476242] [0x00007f0affdc9440] [info]    Checking if indices need to be installed from resources...
Multisample enabled
The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 135 error_code 2 request_code 154 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
`

When running an older version of the slic3r I get the same error. E.g. if I run this one:

./Slic3rPE-1.42.0-beta+linux64-full-201903171406.AppImage

`./Slic3rPE-1.42.0-beta+linux64-full-201903171406.AppImage --loglevel 100
Gtk-Message: 11:09:52.672: Failed to load module "overlay-scrollbar"
[2019-05-30 11:09:52.709414] [0x00007f80e68e2440] [info]    Checking if indices need to be installed from resources...
Multisample enabled
The program 'slic3r' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 131 error_code 2 request_code 154 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

`

vojtechkral, 2019-05-30:
@josbouten Thanks. I wonder if the missing module error message causes the subsequent X error. Could you perhaps test that by reinstalling the scrollbar package as suggested [here](https://askubuntu.com/questions/984556/gtk-message-failed-to-load-module-overlay-scrollbar-ubuntu-17-10/1014525):

    sudo apt-get install --reinstall overlay-scrollbar-gtk2

TheMrCaveman, 2019-06-02:
> @josbouten Thanks. I wonder if the missing module error message causes the subsequent X error. Could you perhaps test that by reinstalling the scrollbar package as suggested [...]

That fixed it for me!

bubnikv, 2019-06-03:
cool, closing then.

closed 2019-06-03 by bubnikv