prusa3d/PrusaSlicer #2399: text in color print window

bhayss, 2019-05-27:
2.0.0-rd+ win64
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

win10
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Mk3
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

color print pop-up window text should read 
above 'layer x'
up to 'layer x'

* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

lukasmatena, 2019-05-28:
Would you please be more specific?

BerndJM, 2019-05-28:
I think he means the "gaps" in the heights:
e.g. _up to 10.20mm -- above 10.40mm_
shouldn't it be _up to 10.20mm -- above 10.20mm_ ?

bhayss, 2019-05-28:
Yes, exactly.
Just a choice of proper words.
"Up to level x" can mean up to and including.
"Above level y" generally does not include level y.

closed 2026-09-29 by prusaslicer-bot[bot]